bblanchon/ArduinoJson #1565: InvalidInput

kunmonster, 2021-05-22:
Hello! 
I'am sorry to ask the question , i use the newest lib ArduinoJson , when i decode the json string "{"ssid":"test","password":"123456"}" , it always show me a message as "InvalidInput" .
then i use the assistant to help me get the code to ESP8266 but it show the message too.
the code like this:

`String input = incomingClient.readString();
//the input is {"ssid":"test","password":"123456"}" , i have used the print to check 
StaticJsonDocument<96> doc;
  DeserializationError error = deserializeJson(doc, input);

  if (error) {
    Serial.print(F("deserializeJson() failed: "));
    Serial.println(error.f_str());
    return;
  }

  const char* ssid = doc["ssid"];          // "k123456"
  const char* password = doc["password"];  // "123456kun"`

bblanchon, 2021-05-23:
Hi @kunmonster,

Please see:

* [`DeserializationError`](https://arduinojson.org/v6/api/misc/deserializationerror/)

Best regards,
Benoit

kunmonster, 2021-05-23:
Hi,
I hava look up the doc again and read the all situation carefully,
exclude the simply invalid and it doesn't contain a comment.

Firstly i think it's possible that the input conatin BOM but after i erase the space in the input , and use the print to check the first character  and it's "{",but it also catch the InvalidInput

[image]

[image]

So,it is very puzzling , wish your reply!

bblanchon, 2021-05-24:
Replacing spaces with an empty string will not remove the BOM because it's not made of spaces (`0x20`) but non-printable characters (`0xFF` and `0xFE`).
Instead, you should skip the two first bytes, like so:

```c++
client.read();
client.read();
deserializeJson(doc, client);
```

I recommend that you print these two bytes, so we can make sure this is really a BOM issue.

kunmonster, 2021-05-30:
I have already find the problem.
The reason why the Invalid is that the WiFiCilent.readString() contains the headers！ It's so puzzling! Thank for your help😊

bblanchon, 2021-05-30:
Indeed, I eliminated this possibility after seeing this picture:

[image]

...which you also confirmed by saying 

> and use the print to check the first character and it's "{"

Next time, please, don't hide critical information.

closed 2021-05-30 by bblanchon