bblanchon/ArduinoJson #1567: strptime not working anymore Feature published in [ArduinoJson 6.18.0](https://arduinojson.org/news/2021/05/04/version-6-18-0/)

ndx1905-github, 2021-05-22:
I was using the strptime function as described here https://arduinojson.org/news/2021/05/04/version-6-18-0/

for some reason, it worked a couple days ago, and now when I try to compile my code, I have an error
'strpftime' was not declared in this scope; did you mean 'strftime'?

I tried to look in the arduinojson 6.18.0 src and found references to strftime, but not to strpftime

I don't know what happened code was working 10 days ago, and now I'm lost

Feature published in [ArduinoJson 6.18.0](https://arduinojson.org/news/2021/05/04/version-6-18-0/)

_Originally posted by @bblanchon in https://github.com/bblanchon/ArduinoJson/issues/687#issuecomment-832930075_

bblanchon, 2021-05-23:
Hi @ndx1905-github,

`strptime()` is a non-standard function, usually provided by the `time.h` header; make sure you include this header.

Did you recently switch to different hardware? Maybe the function is not available on the corresponding core.

Best regards,
Benoit

ndx1905-github, 2021-05-23:
Bonjour Benoit

thanks for the answer.
time.h is indeed included. I've noticed I was using Time.h (capital letter) instead of time.h,  but changing that didn't havee any effect. Also as I said, code was working, and it stopped working without me consciouly changing anything.

but you are right! **i've changed hardware**. I was compiling for a generic ESP8266 module and tried to compile code for a variant of ESP8266, namely a D1 mini.

That was the first time I got an error.

However I switched back to the former setting and hardware, and it doesn't compile anymore for the old hardware which I don't understand at all.

I've also updated the ESP8266 board libraries from 2.7x to 3.0, so maybe that's the problem. I'll try to revert and see if it works

PS: yes strpftime was a typo in this message only and not in the code, sorry

ndx1905-github, 2021-05-23:
That was it..!

code and strptime() compiles fine with version 2.7.4 of ESP8266 board manager https://github.com/esp8266/Arduino

but breaks with version 3.0 that was recently released, for all cards

I've reverted to 2.7.4 and it works fine, but it's annoying not to know why it's not working for 3.0

bblanchon, 2021-05-24:
I think that's a question for ESP8266 core team.
I saw they switched to newlib, which should support `strptime()` as well.

bblanchon, 2021-06-15:
@ndx1905-github, any update?

paulocsanz, 2021-06-21:
This is known upstream: https://github.com/esp8266/Arduino/issues/8122

Just need to add this to the code:

```
extern "C" void *memrchr (const void *__restrict, int __restrict, size_t __restrict);
```

closed 2021-07-03 by bblanchon