bblanchon/ArduinoJson #1568: error: 'const class FirebaseJson' has no member named 'memoryUsage'

RubensOliveira97, 2021-05-23:
Hello guys,

I try to download an code in my ESP32, but i have that problem on platformIO:
[image]
[image]

If need more evidences of problem, ask me, because i'm new in this lib, so i don't know what is necessary for the best analysis.

Thank so much !

bblanchon, 2021-05-23:
Hi @RubensOliveira97,

What on line 49 of main.cpp?
It seems that your creating a `DynamicJsonDocument` from a `FirebaseJson` instance; I don't think that's possible.

Best regards,
Benoit

RubensOliveira97, 2021-05-23:
> Hi @RubensOliveira97,
> 
> What on line 49 of main.cpp?
> It seems that you're creating a `DynamicJsonDocument` from a `FirebaseJson` instance; I don't think that's possible.
> 
> Best regards,
> Benoit

Hello @bblanchon ,

At first, see it:
[image]
The SerialPrint:
[image]

So, the content on line 49 is:
[image]

RubensOliveira97, 2021-05-23:
@bblanchon I found the solution hahah:
[image]
The SerialPrint:
[image]

So, an doubt, what's the best way to split that string in 2 JSON ?

bblanchon, 2021-05-24:
I'm glad you found the solution!
However, I'm not sure how I can help with your second question... can you clarify?

RubensOliveira97, 2021-05-25:
> I'm glad you found the solution!
> However, I'm not sure how I can help with your second question... can you clarify?

So, I have a stringfy json  "{\"-MaBsVgSflB5fhJyEQ9S\":{\"name\":\"GRILL\",\"onOff\":false,\"pin\":\"3\",\"power\":\"600\"},\"-MaLPJj23JV6z4NRFdRx\":{\"name\":\"Geladeira\",\"onOff\":false,\"pin\":\"3\",\"power\":\"300\"}}". What is the more simple form with split  it in two jsons, like this :{"
MaBsVgSflB5fhJyEQ9S":{"name":"GRILL"
"onOff":false
"pin":"3"
"power":"600"}
"
MaLPJj23JV6z4NRFdRx":{"name":"Geladeira"
"onOff":false
"pin":"3"
"power":"300"}}.

Say the ESP don't know what's the keys(MaBsVgSflB5fhJyEQ9S" for example ) in case.if I can separate or json from the firebase, according to the keys, then it is like this to use the split () to change each attribute. I don't know is to separate the JSON into a JSON sub.

bblanchon, 2021-05-28:
I'm sorry, @RubensOliveira97, but I still don't understand what you mean.
I recommend that you start by reading the documentation and maybe come back here if you have any problems.
Start here: [Deserialization tutorial](https://arduinojson.org/v6/doc/deserialization/)

closed 2021-06-15 by bblanchon