bblanchon/ArduinoJson #1570: client:60:1: error: 'doc' does not name a type  doc["type"] = "TEM";

Jesus8000, 2021-05-23:
i'm copying from the documentation : and it dosnt work

StaticJsonDocument<200> doc;

doc["type"] = "TEM";
doc["bootNr"] = 17500;
doc["gewicht"] = 2000;

bblanchon, 2021-05-24:
Hi @Jesus8000,

Can you copy/paste the complete compiler output?

Best regards,
Benoit

Jesus8000, 2021-05-24:
StaticJsonDocument<200> doc; //preallocate the jsonBuffer für jeden eintrag eine ["stockNr"] = 1;

doc["type"] = "TEM";
doc["bootNr"] = 17500;
doc["gewicht"] = 2000;

String serializeJsonOutputString = "JSON = ";
serializeJson(doc, serializeJsonOutputString);
------------------------------------------------------------------

client:60:1: error: 'doc' does not name a type
 doc["type"] = "TEM";
 ^
client:61:1: error: 'doc' does not name a type
 doc["bootNr"] = 17500;
 ^
client:62:1: error: 'doc' does not name a type
 doc["gewicht"] = 2000;
 ^
client:65:14: error: expected constructor, destructor, or type conversion before '(' token
 serializeJson(doc, serializeJsonOutputString);
              ^
Mehrere Bibliotheken wurden für "WiFi.h" gefunden
 Benutzt: C:\Users\<>\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\libraries\WiFi
 Nicht benutzt: C:\Program Files (x86)\Arduino\libraries\WiFi
exit status 1
'doc' does not name a type

bblanchon, 2021-05-24:
Unlike many popular languages like Python or JavaScript, the C++ language doesn't allow statements at the global scope.
Instead, you must move all statements into a function.
The entry point for a regular C++ program is the `main()` function, but in an Arduino program, it's the `setup()` function.

Here is how you could rewrite your program:

```c++
void setup() {
  StaticJsonDocument<200> doc;
  doc["type"] = "TEM";
  doc["bootNr"] = 17500;
  doc["gewicht"] = 2000;

  String serializeJsonOutputString = "JSON = ";
  serializeJson(doc, serializeJsonOutputString);
}
```

See also:

* [JsonGeneratorExample.ino](https://arduinojson.org/v6/example/generator/)

Jesus8000, 2021-05-25:
ah okey thx.

closed 2021-05-26 by bblanchon