bblanchon/ArduinoJson #1571: Avoiding Dangling Pointers in pubsubclient but for int or float

LenardValdez, 2021-05-24:
I'm fairly new to Arduino and certainly in ArduinoJson. As I was reading the Documentation in using PubSubClient It was advised to avoid dangling pointers(always should). but I cant figure out how to do it with storing the doc(the sample json name) value to a struct of an int or float type. 

So here's my callback block and function that it passes the variables to

```
void newCrop(StaticJsonDocument<256> doc,  byte* payload, unsigned int inputLength) {

  DeserializationError error = deserializeJson(doc, payload, inputLength);

  if (error) {
    Serial.print(F("deserializeJson() failed: "));
    Serial.println(error.f_str());
    return;
  }

  tolerance.air_humidity_min = doc["air_humidity"][0]; //float
  tolerance.air_humidity_max = doc["air_humidity"][1];

  tolerance.air_temperature_min = doc["air_temperature"][0]; //float
  tolerance.air_temperature_max = doc["air_temperature"][1];

  tolerance.ec_reading_min = doc["ec_reading"][0];
  tolerance.ec_reading_max = doc["ec_reading"][1];

  tolerance.ph_reading_min = doc["ph_reading"][0];
  tolerance.ph_reading_max = doc["ph_reading"][1];
}

void callback(char* topic, byte* payload, unsigned int length) {
  Serial.print("Message arrived [");
  Serial.print(topic);
  Serial.print("] ");
  Serial.println("");

  //newcrop callback 
  if(!initialized){
    if(!strcmp(topic, command_new_crop)){
      StaticJsonDocument<256> doc;
      newCrop(doc, payload, length);
    }
  }
```
also, the way I pass the whole payload and StaticJsonDocument to a function, is that okay? or there's a better way to do it? Im passing the whole thing because I have several topics that do different things with different json structure.

bblanchon, 2021-05-24:
Hi @LenardValdez,

It seems that you only store `float`s and `int`s in your global structure; I don't see any pointer, so there is no risk of creating a dangling pointer.

However, I recommend that you move the declaration of `doc` inside the `newCrop()` function because, as it's currently written, it calls the function with a temporary copy.
If you absolutely want to have the declaration of `doc` in the `callback()` function, you should change the first parameter of `newCrop()` to be a `JsonDocument&` because it would avoid the temporary copy.

See also #1564.

Best regards,
Benoit

LenardValdez, 2021-05-24:
> Hi @LenardValdez,
>
> It seems that you only store `float`s and `int`s in your global structure; I don't see any pointer, so there is no risk of creating a dangling pointer.
>
> However, I recommend that [...]

I see. Thank you so much for this insight!.

Amazing work with this Library btw!

closed 2021-05-24 by LenardValdez