bblanchon/ArduinoJson #1573: gdb can't debug with `ARDUINOJSON_USE_DOUBLE 1` set

bugrasan, 2021-05-27:
* target platform => linux
* compiler model and version => platformio v5.1.1 (arduino 
* platform => Espressif 8266
* board => WeMos D1 mini Pro
* MVCE (https://stackoverflow.com/help/mcve)

When `ARDUINOJSON_USE_DOUBLE 1` is enabled gdb throws the following error:
```
(gdb) target remote /dev/ttyUSB0
Remote debugging using /dev/ttyUSB0
0x00000020 in ?? ()
Cannot access memory at address 0x20
```
and no debugging is possible. when the line is removed gdb works again.

created a sample proj with platformio and vscode:
https://github.com/bugrasan/esp8266-pio-gdbstub-arduinojson

bblanchon, 2021-05-27:
Hi @bugrasan,

In your program, I see `StaticJsonDocument<4096>`, which is way too big for the ESP8266.
By default, the [stack is limited to 4096 bytes with the ESP8266 core for Arduino](https://github.com/esp8266/Arduino/blob/60fe7b4ca8cdca25366af8a7c0a7b70d32c797f8/cores/esp8266/cont.h#L27).
A stack overflow causes undefined behaviors like, for example, a crash in the debugger.

You must switch to a [`DynamicJsonDocument`](https://arduinojson.org/v6/api/dynamicjsondocument/).

See also:

* [JsonDocument](https://arduinojson.org/v6/api/jsondocument/)
* [Deserialization tutorial](https://arduinojson.org/v6/doc/deserialization/)
* The Reddit case study in [Mastering ArduinoJson](https://arduinojson.org/book/)

Best regards,
Benoit

bugrasan, 2021-05-28:
Thank you, i will consult your book again in more depth :-)

closed 2021-05-28 by bugrasan