bblanchon/ArduinoJson #1575: deserializeJson() return OK is buffer (char *) contains number string

hallard, 2021-05-29:
Benoit

Not sure it's an issue or by concept, migrating to ArduinoJson V6 now I'm using `deserializeJson()` checking error code.

I'm using AsyncMQTT on ESP32 without any issue. But when data received is not JSON I receive `DeserializationError::InvalidInput` except if data (char *) is a number. I'm monitoring my Mosquitto Broker.

Take a look of output 

```
03:26:57.132 -> Topic $SYS/broker/uptime => Value '9614 seconds'
03:26:57.132 -> deserializeJson InvalidInput 
03:26:57.132 -> Topic $SYS/broker/clients/inactive => Value '1'
03:26:57.132 -> deserializeJson OK
03:26:57.132 -> Topic $SYS/broker/clients/disconnected => Value '1'
03:26:57.132 -> deserializeJson OK
```
on topic `$SYS/broker/uptime` I'm receiving `9614 seconds` and `deserializeJson()` return `InvalidInput` which makes sense but

on topic `$SYS/broker/clients/inactive` I'm receiving `1` and deserializeJson()` return `OK`

Did I missed something?

bblanchon, 2021-05-29:
Hi,

A number is a valid JSON document.
You can do the same in your web browser:

```
> JSON.parse('9614 seconds')
Uncaught SyntaxError: Unexpected token s in JSON at position 5
    at JSON.parse (<anonymous>)
    at <anonymous>:1:6
> JSON.parse('1')
< 1
```

Best regards,
Benoit

hallard, 2021-05-29:
thanks was not aware of this. Interesting so my next question So how can I check/detect  with arduino json that it’s just a number? I mean how I access since no field or named value ? May be I should need to double check documentation but don’t remember ever seen something like that.
Thanks

bblanchon, 2021-05-29:
[JsonDocument](https://arduinojson.org/v6/api/jsondocument/) offers an interface similar to [JsonVariant](https://arduinojson.org/v6/api/jsonvariant/):

1. [JsonDocument::is<T>()](https://arduinojson.org/v6/api/jsondocument/is/) tells wether the document is of type `T`.
2. [JsonDocument::as<T>()](https://arduinojson.org/v6/api/jsondocument/as/) casts the document to type `T`

For example, you can do:

```c++
if (doc.is<int>()) {
  int value = doc.as<int>();
  ...
}
```

This feature is rarely used, so it's probably less documented than the rest of the library.
If you have any advice on improving the documentation, I'll be glad to implement it.

hallard, 2021-05-29:
Benoit,

`doc.is<int>()` if what I needed because I need to route management if JSON (in term of object with `{}`) or other string. 

As I even didn't knew that only one number was valid JSON (even if I use JSON everyday for years) this is what I missed.

Thanks and congrats for this great job.

closed 2021-05-29 by hallard

bblanchon, 2021-05-30:
You're welcome, @hallard.
Thank you for using ArduinoJson.
Good luck with your project!