bblanchon/ArduinoJson #1576: Error installing ArduinoJson into PlatformIO

kovachjk, 2021-05-29:


bblanchon, 2021-05-30:
Hi @kovachjk,

There are many users of ArduinoJson on ESP32 with PlatformIO, but that's the first time I hear of such a problem.
The right syntax should be `lib_deps = bblanchon/ArduinoJson@^6.18.0`.
You can also use the PlatformIO command line: `pio lib install "bblanchon/ArduinoJson@^6.18.0"`
Please check the installation instructions in https://platformio.org/lib/show/64/ArduinoJson/installation

Arduino_JSON is another library from another developer.
See a (biased) comparison here: https://arduinojson.org/news/2019/11/19/arduinojson-vs-arduino_json/

Best regards,
Benoit

kovachjk, 2021-05-30:
@bblanchon 
Benoit, Thank you for you attention to your project and my "problem"
For me the fastest way to solve a problem is to go public with it ;-) 
That is why many times I spend hours or days trying to solve it myself.
Once again, this morning I added by project into the project section of the platformio home page and it worked.
While using Platformio for over a year, I have never had to do this, including when I used ArduinoJson. But it had been several revs ago that I used AJ an I had just upgraded PIO and I convinced myself that something broke.  
Again, sorry for the trouble and thank you for being there.  Keep up the good work.

closed 2021-05-30 by kovachjk

bblanchon, 2021-05-30:
No worries. Things like this happen to everyone.
Anyway, thank you for using ArduinoJson, @kovachjk.
Don't forget to [cast a star](https://github.com/bblanchon/ArduinoJson/stargazers) to support the project :wink: