bblanchon/ArduinoJson #1578: Add std::string_view support

paulocsanz, 2021-06-03:
We heavily use non-zero-terminated std::string_view's through our code, but we are not able to serialize them.

Support for std::string_view was mentioned in #704 as something of interest for this lib, but it seems the discussion ended with the stream solution. It's not clear to me how to use it, I will understand it better, but it seems official support for std::string_view still would be very useful.

Are there plans for direct support?

bblanchon, 2021-06-04:
Hi @paulocsanz,

PR #1554 should already provide partial support for `std::string_view`, including `operator[]`.
I plan on adding the complete support for `std::string_view` soon, possibly in v6.18.1.

Best regards,
Benoit

closed 2021-06-18 by bblanchon

bblanchon, 2021-07-03:
This feature is available in [ArduinoJson 6.18.1](https://github.com/bblanchon/ArduinoJson/releases/tag/v6.18.1)

commit 821bf70d (2021-06-16): Added support for std::string_view (closes #1578, closes #1554)
commit 816b8d0a (2021-06-17): Added support for std::string_view (closes #1578, closes #1554)
commit f2351574 (2021-06-17): Added support for std::string_view (closes #1578, closes #1554)