bblanchon/ArduinoJson #1580: Default empty filter (equivalent to no filter at all) for function argument

Kuenlun, 2021-06-05:
Hi, I have a very simple question.
Is it possible to create an empty filter so that no filter is done?
I'm trying to create a filter that does nothing because I have a function that sometimes requires a filter and sometimes not. If I could pass a default empty filter it would simplify my code and eliminate the need for function overloading (in my case, code duplication).
While using [](https://arduinojson.org/v6/assistant/) I found this:
```
StaticJsonDocument<0> filter;
filter.set(true);
```
But how do I use it as a default function argument?
Thanks for your awesome project.

bblanchon, 2021-06-07:
Hi @Kuenlun,

You will have to make the default filter document global, and set the content in the program's entry point, like so:

```c++
StaticJsonDocument<0> defaultFilter;

DeserializationError parseJson(JsonDocument& doc, const char* input, const JsonDocument& filter=defaultFilter) {
  return deserializeJson(doc, input, DeserializationOption::Filter(filter));
}

void setup() {
  defaultFilter.set(true);
}
```

I wish we could set the content of the filter in the constructor (as [nlohmann/json](https://github.com/nlohmann/json) does), but ArduinoJson doesn't support it yet, and I don't plan to add this feature in the foreseeable future.

Best regards,
Benoit

Kuenlun, 2021-06-07:
I assumed it was not supported yet, but wanted to confirm.
Thanks for your quick response.

closed 2021-06-07 by Kuenlun

bblanchon, 2021-06-07:
You're welcome @Kuenlun; thank you for using ArduinoJson!