bblanchon/ArduinoJson #1581: Unable to copy char* from JSON field

baconcheese113, 2021-06-07:
Thanks for the awesome library! I'm currently trying to parse a small GraphQL response and copy the token I receive to a class member. Sometimes (50% of the time) the `char* token` is assigned the whole string properly, but the `network.accessToken` copy almost always only copies a substring of the original char*.

 - Arduino Nano 33 IoT
 - atmelsam@6.0.1
 - PlatformIO/VSCode
 - ArduinoJson@^6.18.0

An example of the longest graphql response `{\"data\": {\"loginAsHub\":\"eyJhbGciOiJIUzI1NiJ9.SHViOjEy.eaI-PksPs0UZ16MzGhD7biOYyRdIVb2cms_yzy--ZFQ\"}}`

This is the basic code
```c++
int32_t userIdValue = 1;
char mutationStr[100 + sizeof userIdValue + strlen(DEVICE_SERIAL)]{};
snprintf(mutationStr, sizeof(mutationStr), "{\"query\":\"mutation loginAsHub{loginAsHub(userId:%ld, serial:\\\"%s\\\")}\",\"variables\":{}}\n", userIdValue, DEVICE_SERIAL);
StaticJsonDocument<400> doc = network.SendRequest(mutationStr);
if(doc["data"] && doc["data"]["loginAsHub"]) {
  const char* token = (const char*)(doc["data"]["loginAsHub"]);
  Serial.print("token is: ");
  Serial.println(token);
  Serial.print("token length is: ");
  Serial.println(strlen(token));
  strncpy(network.accessToken, token, strlen(token));
  Serial.print("accessToken is: ");
  Serial.println(network.accessToken);
  Serial.print("accessToken length is: ");
  Serial.println(strlen(network.accessToken));
} else {
  Serial.println("error parsing doc");
}
Serial.print("FreeMemory: ");
Serial.println(freeMemory());
```

And Network.h accessToken:
```c++
char accessToken[100]{};
```

Network.cpp SendRequest fn https://github.com/baconcheese113/handle-it/blob/master/src/hub/Network.cpp#L8

And the logs
```
Response is: {"data":{"loginAsHub":"eyJhbGciOiJIUzI1NiJ9.SHViOjEy.eaI-PksPs0UZ16MzGhD7biOYyRdIVb2cms_yzy--ZFQ"}}
token is: eyJhbGciOiJIUzI1NiJ9.SHViOjEy.eaI-PksPs0UZ16MzGhD7biOYyRdIVb2cms_yzy--ZFQ
token length is: 34
accesstoken size is: 100
accessToken is: eyJhbGciOiJIUzI1NiJ9.SHViOjEy.ea�
accessToken length is: 34
FreeMemory: 25571
```

Any ideas about what could be the issue here?

bblanchon, 2021-06-08:
Hi @baconcheese113,

When you pass a writeable buffer as the input of [`deserializeJson()`](https://arduinojson.org/v6/api/json/deserializejson/), ArduinoJson uses the zero-copy mode. Instead of copying the strings from the input into the [`JsonDocument`](https://arduinojson.org/v6/api/jsondocument/), it stores pointers to the input buffer.
Therefore, the input buffer must stay alive during the whole lifetime of the [`JsonDocument`](https://arduinojson.org/v6/api/jsondocument/); otherwise, all the strings in the [`JsonDocument`](https://arduinojson.org/v6/api/jsondocument/) will be dangling pointers.

The easiest solution is to disable the zero-copy mode by passing a read-only input buffer. Casting the pointer will do the trick:

```c++
// replace this:
deserializeJson(doc, input, inputSize);

// ...with this:
deserializeJson(doc, (const char*)input, inputSize);
```

Now that the zero-copy mode is disabled, you probably need a larger [`JsonDocument`](https://arduinojson.org/v6/api/jsondocument/), so don't forget to update the capacity. As usual, use the [ArduinoJson Assistant](https://arduinojson.org/v6/assistant/) to compute the right capacity for your project.

Best regards,
Benoit

PS: answer taken from the new [ArduinoJson Troubleshooter](https://arduinojson.org/v6/troubleshooter/).

baconcheese113, 2021-06-08:
Appreciate the thorough response Benoit! I was using the assistant before, but didn't know about that troubleshooter. It's fantastic!

From what I understand the issue was because I was declaring the `StaticJsonDocument` in scope local to the function, and then trying to return the pointer to it and access it from outside of the function?
```c++
StaticJsonDocument<400> doc;
DeserializationError error = deserializeJson(doc, response);
...
return doc;
```

In that case (to keep my code aligned with best practices), should I be passing in a reference to the doc from where I call it instead? Something like
```c++
StaticJsonDocument<400> doc;
network.SendRequest(doc, mutationStr);

// definition of SendRequest
void Network::SendRequest(StaticJsonDocument<400> doc, char* query) {
```

bblanchon, 2021-06-09:
No, the scope of the `JsonDocument` is not the problem.
It's the [char response[200]](https://github.com/baconcheese113/handle-it/blob/master/src/hub/Network.cpp#L39), which is local to `Network::SendRequest()`, that is the cause of your issue.

Still, it's a good idea to move the `StaticJsonDocument` out of the function to avoid the copy (unless RVO kicks in, but there is no guarantee). Alternatively, you can use `DynamicJsonDocument` which support move semantics, and therefore, doesn't copy the memory pool.

baconcheese113, 2021-06-11:
I ended up going with 
```c++
DynamicJsonDocument Network::SendRequest(char* query) {
    ...
    DynamicJsonDocument doc(400);
    DeserializationError error = deserializeJson(doc, (const char*)response);
    ...
    return doc;
}
```
And calling it like
```c++
DynamicJsonDocument loginDoc = network.SendRequest(loginMutationStr);
```

bblanchon, 2021-06-11:
This looks good 👍 
Does it work as expected?

baconcheese113, 2021-06-11:
Sure does! One other question, if I were to declare `response` as `const char response[400]{}` instead of `char response[400]{}` would I still need the cast in the `deserializeJson` call?

bblanchon, 2021-06-13:
The program would not compile with `const char response[400]{}`
If you don't want to see the cast in your code, simply extract a function that takes a `const char*` as a parameter.

closed 2021-06-16 by baconcheese113