bblanchon/ArduinoJson #1582: unable to compile for platformIO on macOS: attributes are not allowed on a function-definition

jessesteinen, 2021-06-07:
Hi,

I'm compiling some hobby project (cityair.os, https://github.com/appsmonkey/air.nodemcu.full) which includes the ArduinoJson library (I'm building this using Atom and Platform IO, latest versions, for NodeMCU v2).

However I'm getting the following error (full build output, I think the code needs touching up anyway, but this is the only breaking error so far):
`.pio/libdeps/nodemcuv2/ArduinoJson/src/ArduinoJson/MsgPack/MsgPackSerializer.hpp:28:7: error: attributes are not allowed on a function-definition
   28 |       ARDUINOJSON_NO_SANITIZE("float-cast-overflow") {
      |       ^~~~~~~~~~~~~~~~~~~~~~~`

Any help would be much appreciated. Many thanks for creating this library in the first place, Benoit!

Thx,
Jesse

`Jesses-MBP:air.nodemcu.full jesse$ platformio run
Processing nodemcuv2 (platform: espressif8266; board: nodemcuv2; framework: arduino)
--------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/nodemcuv2.html
PLATFORM: Espressif 8266 (3.0.0) > NodeMCU 1.0 (ESP-12E Module)
HARDWARE: ESP8266 160MHz, 80KB RAM, 4MB Flash
PACKAGES: 
 - framework-arduinoespressif8266 3.30000.210519 (3.0.0) 
 - tool-esptool 1.413.0 (4.13) 
 - tool-esptoolpy 1.30000.201119 (3.0.0) 
 - toolchain-xtensa 2.100200.0 (10.2.0)
Converting Air.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 78 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <Adafruit Unified Sensor> 1.0.3
|-- <DHT sensor library> 1.3.4
|   |-- <Adafruit Unified Sensor> 1.0.3
|-- <EspSoftwareSerial> 6.12.3
|-- <Adafruit NeoPixel> 1.8.2
|-- <Adafruit BME280 Library> 2.1.4+sha.ee53e8b
|   |-- <Adafruit Unified Sensor> 1.0.3
|   |-- <SPI> 1.0
|   |-- <Wire> 1.0
|-- <SparkFun CCS811 Arduino Library> 2.0.1+sha.6d00405
|   |-- <Wire> 1.0
|-- <BH1750> 1.2.0+sha.37068ca
|   |-- <Wire> 1.0
|-- <Time> 1.6.0+sha.410aee4
|-- <paulvha SCD30 Library> 1.0.4+sha.6afe3e6
|   |-- <Wire> 1.0
|-- <Wire> 1.0
|-- <SPI> 1.0
|-- <OneWire> 2.3.4
|-- <ArduinoJson> 6.10.0
|-- <NTPClient> 3.1.0
|-- <Device>
|   |-- <BME_280>
|   |   |-- <CityOS>
|   |   |   |-- <Util>
|   |   |   |   |-- <Adafruit NeoPixel> 1.8.2
|   |   |   |-- <ArduinoJson> 6.10.0
|   |   |   |-- <AWS>
|   |   |   |   |-- <Util>
|   |   |   |   |   |-- <Adafruit NeoPixel> 1.8.2
|   |   |   |   |-- <ArduinoJson> 6.10.0
|   |   |   |   |-- <Config>
|   |   |   |   |   |-- <Util>
|   |   |   |   |   |   |-- <Adafruit NeoPixel> 1.8.2
|   |   |   |   |   |-- <ArduinoJson> 6.10.0
|   |   |   |   |-- <NTPClient> 3.1.0
|   |   |   |   |-- <PubSubClient> 2.7
|   |   |   |   |-- <ESP8266WiFi> 1.0
[... 186 lines cut ...]
In file included from lib/Config/Config.h:16,
                 from lib/Config/Config.cpp:1:
/Users/jesse/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/FS.h:275:15: note: declared here
  275 | extern fs::FS SPIFFS __attribute__((deprecated("SPIFFS has been deprecated. Please consider moving to LittleFS or other filesystems.")));
      |               ^~~~~~
lib/Config/Config.cpp:9:29: warning: 'SPIFFS' is deprecated: SPIFFS has been deprecated. Please consider moving to LittleFS or other filesystems. [-Wdeprecated-declarations]
    9 |             File configFile=SPIFFS.open(F("/config.json"), "r");
      |                             ^~~~~~
In file included from lib/Config/Config.h:16,
                 from lib/Config/Config.cpp:1:
/Users/jesse/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/FS.h:275:15: note: declared here
  275 | extern fs::FS SPIFFS __attribute__((deprecated("SPIFFS has been deprecated. Please consider moving to LittleFS or other filesystems.")));
      |               ^~~~~~
Compiling .pio/build/nodemcuv2/lib296/ESP8266WiFi/ESP8266WiFiSTA-WPS.cpp.o
Archiving .pio/build/nodemcuv2/libdad/libPubSubClient.a
Indexing .pio/build/nodemcuv2/libdad/libPubSubClient.a
Compiling .pio/build/nodemcuv2/lib296/ESP8266WiFi/ESP8266WiFiSTA.cpp.o
*** [.pio/build/nodemcuv2/lib230/Config/Config.cpp.o] Error 1
*** [.pio/build/nodemcuv2/src/Air.ino.cpp.o] Error 1
========================== [FAILED] Took 7.61 seconds ==========================

Environment    Status    Duration
-------------  --------  ------------
nodemcuv2      FAILED    00:00:07.607
==================== 1 failed, 0 succeeded in 00:00:07.607 ====================`

bblanchon, 2021-06-08:
Hi @jessesteinen,

This bug was fix in 9862048a58a81e45f6d4468a713ede099644c123, which was first published in v6.10.1 two years ago.
Please upgrade ArduinoJson.

Best regards,
Benoit

jessesteinen, 2021-06-10:
Many thanks, Benoit. Somehow I must have missed that while searching this github for the error that I got.

closed 2021-06-10 by bblanchon