bblanchon/ArduinoJson #1583: Move from ArduinoJson 5 to ArduinoJson 6

miloit, 2021-06-08:
Hello,
I have aproblem to reprogramm my code

```
DynamicJsonBuffer jsonBuffer;
    JsonObject& json = jsonBuffer.createObject();
json["mqtt_server"] = mqtt_server;

    File configFile = SPIFFS.open("/config.json", "w");
    if (!configFile) {
      Serial.println("failed to open config file for writing");
    }

    json.printTo(Serial);
    json.printTo(configFile);
    configFile.close();
```

to the new Arduinojson Class 6.

Can you please help me?

bblanchon, 2021-06-08:
Hi @miloit,

Here is the equivalent for ArduinoJson 6:

```c++
DynamicJsonDocument doc(512);
doc["mqtt_server"] = mqtt_server;

File configFile = SPIFFS.open("/config.json", "w");
if (!configFile) {
  Serial.println("failed to open config file for writing");
}

serializeJson(doc, Serial);
serializeJson(doc, configFile);
configFile.close();
``` 

Should you need any information, please refer to:

* [Migrating from version 5 to 6](https://arduinojson.org/v6/doc/upgrade/)
* [JsonConfigFile.ino](https://arduinojson.org/v6/example/config/)
* [Serialization Tutorial](https://arduinojson.org/v6/doc/serialization/)
* The SPIFFS case study in [Mastering ArduinoJson](https://arduinojson.org/book/)

Best regards,
Benoit

miloit, 2021-06-09:
and this part ? 

`DynamicJsonBuffer jsonBuffer;
        JsonObject& json = jsonBuffer.parseObject(buf.get());
        json.printTo(Serial);
        if (json.success()) {
          Serial.println("\nparsed json");
          strcpy(deviceName, json["deviceName"]);
          strcpy(mqttServer, json["mqttServer"]);
          strcpy(mqttPort, json["mqttPort"]);
          strcpy(mqttUser, json["mqttUser"]);
          strcpy(mqttPassword, json["mqttPassword"]);
        } else {
          Serial.println("failed to load json config");
        }`

bblanchon, 2021-06-10:
Check the links I sent in my previous message.

closed 2021-07-03 by bblanchon