bblanchon/ArduinoJson #1585: error: inlining failed in call to 'always_inline'

lbussy, 2021-06-10:
I'm getting this error in an application that used to compile without issue on an older ArduinoJson 6.x version.

<details><summary>ArduinoJson Troubleshooter's report</summary> <ol><li>The issue happens at compile time</li><li>The error is not in the list</li></ol></details>

```
In file included from .pio\libdeps\d1_mini\ArduinoJson\src/ArduinoJson/Array/ArrayIterator.hpp:8,
                 from .pio\libdeps\d1_mini\ArduinoJson\src/ArduinoJson/Array/ArrayRef.hpp:8,
                 from .pio\libdeps\d1_mini\ArduinoJson\src/ArduinoJson.hpp:17,
                 from .pio\libdeps\d1_mini\ArduinoJson\src/ArduinoJson.h:9,
                 from src\jsonconfig.h:27,
                 from src\jsonconfig.cpp:23:
.pio\libdeps\d1_mini\ArduinoJson\src/ArduinoJson/Variant/VariantRef.hpp: In member function 'typename ArduinoJson6180_D1::enable_if<((! ArduinoJson6180_D1::is_same<T, char*>::value) && (! ArduinoJson6180_D1::is_same<T, char>::value)), T>::type ArduinoJson6180_D1::VariantConstRef::as() const [with T = ArduinoJson6180_D1::VariantRef]':
.pio\libdeps\d1_mini\ArduinoJson\src/ArduinoJson/Variant/VariantRef.hpp:295:16: error: inlining failed in call to 'always_inline' 'ArduinoJson6180_D1::VariantConstRef::operator T() const [with T = ArduinoJson6180_D1::VariantRef]': function not inlinable
  295 |   FORCE_INLINE operator T() const {
      |                ^~~~~~~~
.pio\libdeps\d1_mini\ArduinoJson\src/ArduinoJson/Variant/VariantRef.hpp:253:34: note: called from here
  253 |     return Converter<T>::fromJson(*this);
      |            ~~~~~~~~~~~~~~~~~~~~~~^~~~~~~
Compiling .pio\build\d1_mini\src\uptime.cpp.o
*** [.pio\build\d1_mini\src\jsonconfig.cpp.o] Error 1
```

I'm more than happy to try to refine the issue, create a small fail case, etc., however, to rip it all out and re-write it would sort of remove any advantage your experience might bring. :)

If this error rings a bell with you, great!  If not, I'll start commenting out swaths of code to figure out where I'm going sideways.

```
Updating bblanchon/ArduinoJson                6.18.0                             [Up-to-date]

Platform espressif8266
--------
Updating platformio/espressif8266             3.0.0                              [Up-to-date]
Updating platformio/toolchain-xtensa          2.100200.0 @ ~2.100200.0           [Up-to-date]
Updating platformio/framework-arduinoespressif8266 3.30000.210519 @ ~3.30000.0        [Up-to-date]
```

bblanchon, 2021-06-10:
Hi @lbussy,

Thank you very much for reporting this error 👍 
I currently don't manage to reproduce it.
Can you show me the relevant lines from `jsonconfig.cpp`?

Best regards,
Benoit

lbussy, 2021-06-10:
I can/will - I was hoping you'd say "yeah you did X dummy!" and fix it. :)

I'm going to have to carve out a bunch of things to make a specific test to narrow it down.  The errors I get always seem to point to the header that included ArduinoJson and rarely to the line causing the exception.

lbussy, 2021-06-10:
Okay, found it.  This is the functionality you helped me with previously.  It's possible I made it work in a manner that was not intended and a patch broke me:

``` cpp
bool mergeJsonString(String newJson)
{
    DynamicJsonDocument doc(capacityDeserial);
    DeserializationError err = deserializeJson(doc, newJson);
    if (err)
        Serial.println(err.c_str());
    return mergeJsonObject(doc);
}

bool mergeJsonObject(JsonVariantConst src)
{
    DynamicJsonDocument doc(capacityDeserial);
    JsonObject root = doc.to<JsonObject>();
    config.save(root);
    if (merge(root, src))
    {
        config.load(root);
        saveFile();
        return true;
    }
    return false;
}

bool merge(JsonVariant dst, JsonVariantConst src)
{
    if (src.is<JsonObject>())
    {
        for (auto kvp : src.as<JsonObject>())
        {
            merge(dst.getOrAddMember(kvp.key()), kvp.value());
        }
    }
    else
    {
        dst.set(src);
    }
    return true;
}
```

I'm relatively certain it's in the `bool merge(JsonVariant dst, JsonVariantConst src)` function, although I am not certain why.

bblanchon, 2021-06-11:
You did make a mistake, but it wasn't easy to spot!

`src.as<JsonObject>()` is not a valid conversion because `src` is a `JsonVariantConst`.
Indeed, you cannot get a read-write reference out of a read-only reference.
Instead, you must use `src.as<JsonObjectConst>()`.

I added the dummy class `InvalidConversion` (declared but never defined) to get a more explicit compilation error:

```
In file included from .../ArduinoJson/Array/ArrayIterator.hpp:8:0,
                 from .../ArduinoJson/Array/ArrayRef.hpp:8,
                 from .../ArduinoJson.hpp:17,
                 from .../ArduinoJson.h:9,
                 from MyProgram.ino:9:
.../ArduinoJson/Variant/VariantRef.hpp: In instantiation of 'typename ArduinoJson6180_91::enable_if<((! ArduinoJson6180_91::is_same<T, char*>::value) && (! ArduinoJson6180_91::is_same<T, char>::value)), T>::type ArduinoJson6180_91::VariantConstRef::as() const [with T = ArduinoJson6180_91::ObjectRef; typename ArduinoJson6180_91::enable_if<((! ArduinoJson6180_91::is_same<T, char*>::value) && (! ArduinoJson6180_91::is_same<T, char>::value)), T>::type = ArduinoJson6180_91::ObjectRef]':
MyProgram.ino:37:44:   required from here
.../ArduinoJson/Variant/VariantRef.hpp:251:34: error: invalid use of incomplete type 'class ArduinoJson6180_91::InvalidConversion<ArduinoJson6180_91::VariantConstRef, ArduinoJson6180_91::ObjectRef>'
     return Converter<T>::fromJson(*this);
            ~~~~~~~~~~~~~~~~~~~~~~^~~~~~~
In file included from .../ArduinoJson/Variant/VariantRef.hpp:14:0,
                 from .../ArduinoJson/Array/ArrayIterator.hpp:8,
                 from .../ArduinoJson/Array/ArrayRef.hpp:8,
                 from .../ArduinoJson.hpp:17,
                 from .../ArduinoJson.h:9,
                 from MyProgram.ino:9:
.../ArduinoJson/Variant/Converter.hpp:14:7: note: declaration of 'class ArduinoJson6180_91::InvalidConversion<ArduinoJson6180_91::VariantConstRef, ArduinoJson6180_91::ObjectRef>'
 class InvalidConversion;  // Error here? See https://arduinojson.org/v6/invalid-conversion/
       ^~~~~~~~~~~~~~~~~
```

lbussy, 2021-06-11:
Ahhhhhhhhhhhhhhhhhh!  I'm not sure how you keep it all straight!  :)

Thank you very much for pointing that out!!  By the way that link to https://arduinojson.org/v6/invalid-conversion/ is a 404.

closed 2021-06-11 by lbussy

bblanchon, 2021-07-03:
`InvalidConversion` is available in [ArduinoJson 6.18.1](https://github.com/bblanchon/ArduinoJson/releases/tag/v6.18.1)

commit 1d24caf0 (2021-06-15): Added InvalidConversion to identify invalid conversions (closes #1585)