bblanchon/ArduinoJson #1588: Lists of key/values or dictionaries, how to?

deponovo, 2021-06-11:
Maybe I am missing something, but I could not yet figure out how I could parse/use the following structure:
```
{
    "wifi": {
        [
            "ssid": "ssid1",
            "password": "password for ssid1"
        ],
        [
            "ssid": "ssid2",
            "password": "password for ssid2"
        ] 
    }
}
```

What I tried:
```
DynamicJsonDocument ConfigJson(1024);
deserializeJson(ConfigJson, <json_contents>);
auto ssid1 = ConfigJson["wifi"][0]["ssid"].as<const char*>();
```

bblanchon, 2021-06-12:
Hi @deponovo,

Neither of these examples is a valid JSON document.
Please see an introductory course on JSON.

For example, I created the following material:

* [Serialization and JSON on Arduino](https://www.youtube.com/watch?v=hAB4TdX8dwM) (YouTub video)
* [Mastering ArduinoJson](https://arduinojson.org/book/), chapter 1

Then, of course, there are many other resources on the internet.

Best regards,
Benoit

deponovo, 2021-06-12:
You are right. I think I was working too late in the night -.-
This dict should do it:
```
{
    "wifi":
        [
            {
                "ssid": "ssid1",
                "password": "password for ssid1"
            },
            {
                "ssid": "ssid2",
                "password": "password for ssid2"
            }
        ]
}
```

bblanchon, 2021-06-13:
Please try the [ArduinoJson Assistant](https://arduinojson.org/v6/assistant/); it will write the code for you.

You can find all the information you need in the [Deserialization Tutorial](https://arduinojson.org/v6/doc/deserialization/).

closed 2021-06-15 by deponovo