bblanchon/ArduinoJson #1589: Feature Request: Basic Serial1 (UART) example using Tx and RX

hpssjellis, 2021-06-16:
This looks like a really useful library. I have read a few of the issues and there seems like lots of ways to make data transfer more efficient. Any chance you could show code example for a simple transfer of JSON using Serial1 from one arduino to another, or in my case from one core of the [Arduino Portenta](https://store.arduino.cc/usa/portenta-h7) to the other core.

A simple json string serialized and sent over Serial1 then deserialized so that a few variables can be read.

The  MsgPackParser example looks close for receiving over Serial1  

[examples/MsgPackParser/MsgPackParser.ino](https://github.com/bblanchon/ArduinoJson/blob/6.x/examples/MsgPackParser/MsgPackParser.ino)

and the 

[examples/JsonParserExample/JsonParserExample.ino](https://github.com/bblanchon/ArduinoJson/blob/6.x/examples/JsonParserExample/JsonParserExample.ino)  
looks like it would parse the JSON, but not really sure how the Serial1 would send and receive the information in an efficient way.

I can mess around with it just to see if it sort of works.

bblanchon, 2021-06-17:
Hi @hpssjellis,

Thank you for your interest in ArduinoJson.
Did you check the following page?

* [How to do serial communication between two boards?](https://arduinojson.org/v6/how-to/do-serial-communication-between-two-boards/)

Best regards,
Benoit

hpssjellis, 2021-06-17:
@bblanchon 

Got any sender/receiver examples that don't use SoftwareSerial, that just use Serial1 TX, RX pins. I am loading a bunch of libraries to find the best/easiest way to send variables between cores/boards. 

The best library I have found so far is [SerialTransfer](https://github.com/PowerBroker2/SerialTransfer) but your library looks to have a lot of promise as I am coming from a web/javascript background and have used JSON for several years.

Any suggestions for running the sender/receiver code but with just Serial1?

bblanchon, 2021-06-18:
@hpssjellis, you can easily adapt the example that uses `SoftwareSerial` to `Serial1`.
Very few changes are required: 

* replace `linkSerial` with `Serial1`
* remove the lines specific to `SoftwareSerial`

hpssjellis, 2021-06-18:
@bblanchon . 

Yes changing ```SoftwareSerial``` to ```Serial1``` was reasonably easy. I put the send and receive code onto 2  different  Portenta's using the M7 outer core and  Rx to Tx and GND to GND  I was able to send data:

[image]

But when I put the send code onto the M4 inner core and tried to send the data to the M7 outer core I get this error.

```deserializeJson() returned InvalidInput```

Note: The M7 and M4 cores of the Portenta can't communicate unless you use RPC (remote procedural calls), which works fine but is fairly slow. Topic [here](https://forum.arduino.cc/t/passing-variable-from-cm7-to-cm4-using-rpc/692026/14) on the arduino forum. 

I will file an issue with Arduino mbed core as I don't think this is the fault of your library, unless you have any suggestions:

M4 send code (Note: M4 can't serial print so those lines are commented out)
```
#include <ArduinoJson.h>
//#include <SoftwareSerial.h>

// Declare the "link" serial port
// Please see SoftwareSerial library for detail
//SoftwareSerial linkSerial(10, 11); // RX, TX

void setup() {
  // Initialize "debug" serial port
  // The data rate must be much higher than the "link" serial port
  //Serial.begin(115200);
  //while (!Serial) continue;   // blocking
  //delay(15000);
  //Serial.println("Test Sending ArduinoJSON, using UART TX RX between t2 Portenta M7 cores");

  // Initialize the "link" serial port
  // Use the lowest possible data rate to reduce error ratio
  Serial1.begin(4800);
}
 
void loop() {
  // Values we want to transmit
  long timestamp = millis();
  int value = analogRead(A1);   // A0 is buggy

  // Print the values on the "debug" serial port
 // Serial.print("timestamp = ");
  //Serial.println(timestamp);
 // Serial.print("value = ");
 // Serial.println(value);
 // Serial.println("---");

  // Create the JSON document
  StaticJsonDocument<200> doc;
  doc["timestamp"] = timestamp;
  doc["value"] = value;

  // Send the JSON document over the "link" serial port
  serializeJson(doc, Serial1);

  // Wait
  delay(500);
}

```

M7 receive code (Tx and Rx pins connected)

```
#include <ArduinoJson.h>
//#include <SoftwareSerial.h>

// Declare the "link" serial port
// Please see SoftwareSerial library for detail
//SoftwareSerial linkSerial(10, 11); // RX, TX

void setup() {
  bootM4();
  // Initialize "debug" serial port
  // The data rate must be much higher than the "link" serial port
  Serial.begin(115200);
 // while (!Serial) continue; //blocking
  delay(15000);
  Serial.print("Test Receive ArduoJSON receive UART RX TX between 2 Portenta M7 cores");

  // Initialize the "link" serial port
  // Use the lowest possible data rate to reduce error ratio
  Serial1.begin(4800);
}
 
void loop() {
  // Check if the other Arduino is transmitting
  if (Serial1.available()) 
  {
    // Allocate the JSON document
    // This one must be bigger than for the sender because it must store the strings
    StaticJsonDocument<300> doc;

    // Read the JSON document from the "link" serial port
    DeserializationError err = deserializeJson(doc, Serial1);

    if (err == DeserializationError::Ok) 
    {
      // Print the values
      // (we must use as<T>() to resolve the ambiguity)
      Serial.print("timestamp = ");
      Serial.println(doc["timestamp"].as<long>());
      Serial.print("value = ");
      Serial.println(doc["value"].as<int>());
    } 
    else 
    {
      // Print error to the "debug" serial port
      Serial.print("deserializeJson() returned ");
      Serial.println(err.c_str());
  
      // Flush all bytes in the "link" serial port buffer
      // Is this needed ??
      while (Serial1.available() > 0)
        Serial1.read();
    }
  }
}

```

hpssjellis, 2021-06-19:
@bblanchon Getting closer:

I can do 2 different UART channels, which is better than nothing 
```
/* Portenta Serial UART

// Need to connect pin TX D14 and RX D13 together.

UART myUART0(PA_0,  PI_9,  NC, NC); //TX, RX, RTS, CTC  NOTE: NC means not connected
UART myUART1(PA_9,  PA_10, NC, NC);
UART myUART2(PG_14, PG_9,  NC, NC);
UART myUART3(PJ_8,  PJ_9,  NC, NC);

   //  or
   // Serial
   // Serial0
   // Serial1   is myUART1
   // Serial2
   // Serial3

*/
#include <Arduino.h>

//////////////////// Start All Core M7 Programing /////////////////////
#ifdef CORE_CM7 

#include <ArduinoJson.h>

UART myUART0(PA_0,  PI_9,  NC, NC); //TX, RX, RTS, CTC  NOTE: NC means not connected
UART myUART1(PA_9,  PA_10, NC, NC);

void setup() { 
   bootM4();    
   Serial.begin(115200);
   myUART1.begin(115200); 
   delay(15000);
   Serial.print("Test Receive ArduoJSON receive UART RX TX between 2 Portenta M7 cores");
  
}

void loop() {
  // Check if the other Arduino is transmitting
  if (myUART1.available()) 
  {
    // Allocate the JSON document
    // This one must be bigger than for the sender because it must store the strings
    StaticJsonDocument<300> doc;

    // Read the JSON document from the "link" serial port
    DeserializationError err = deserializeJson(doc, myUART1);

    if (err == DeserializationError::Ok) 
    {
      // Print the values
      // (we must use as<T>() to resolve the ambiguity)
      Serial.print("timestamp = ");
      Serial.println(doc["timestamp"].as<long>());
      Serial.print("value = ");
      Serial.println(doc["value"].as<int>());
    } 
    else 
    {
      // Print error to the "debug" serial port
      Serial.print("deserializeJson() returned ");
      Serial.println(err.c_str());
  
      // Flush all bytes in the "link" serial port buffer
      // Is this needed ??
      while (myUART1.available() > 0)
        myUART1.read();
    }
  }
}

#endif

//////////////////// End All Core M7 Programing /////////////////////

//////////////////// Start All Core M4 Programing /////////////////////

#ifdef CORE_CM4 
#include <ArduinoJson.h>
int  myCount=48;  //48 = ascii 0,    58 ascii 9

UART myUART0(PA_0,  PI_9,  NC, NC); //TX, RX, RTS, CTC  NOTE: NC means not connected
UART myUART1(PA_9,  PA_10, NC, NC);

void setup() { 
    pinMode(LEDB, OUTPUT);   // LEDB = blue, LEDG or LED_BUILTIN = green, LEDR = red 
   // Serial.begin(115200);  // no serial monitor on M4 core without RPC
    myUART0.begin(115200);   

}

void loop() {
   // Values we want to transmit
  long timestamp = millis();
  int value = analogRead(A1);   // A0 is buggy

  // Print the values on the "debug" serial port
 // Serial.print("timestamp = ");
  //Serial.println(timestamp);
 // Serial.print("value = ");
 // Serial.println(value);
 // Serial.println("---");

  // Create the JSON document
  StaticJsonDocument<200> doc;
  doc["timestamp"] = timestamp;
  doc["value"] = value;

  // Send the JSON document over the "link" serial port
  serializeJson(doc, myUART0);

  // Wait
  delay(500);
}

#endif

//////////////////// End All Core M4 Programing /////////////////////
```

bblanchon, 2021-06-19:
Hi @hpssjellis,

I'm glad to see that you're making good progress 👍 

Unfortunately, I have no experience with the Arduino Portenta, so I cannot help you with this.
If I were you, I would first try to transmit one character without using ArduinoJson, just to make sure that the two cores can communicate. When they do communicate, ArduinoJson will work.

Regarding the [`InvalidInput`](https://arduinojson.org/v6/api/misc/deserializationerror/#invalidinput), you can read the troubleshooting section in [How to do serial communication between two boards?](https://arduinojson.org/v6/how-to/do-serial-communication-between-two-boards/) or try the [ArduinoJson Troubleshooter](https://arduinojson.org/v6/troubleshooter/).

Best regards,
Benoit

closed 2021-07-03 by bblanchon