bblanchon/ArduinoJson #1590: Error getting location on NodeMcu ESP8266-12E using unwiredlabs api

RohanMoj1999, 2021-06-16:


RohanMoj1999, 2021-06-16:
I was trying to get the location of NodeMcu using unwiredlabs api. I used the below code:-
`#include <ESP8266HTTPClient.h>
#include  <ArduinoJson.h>
#include "ESP8266WiFi.h"
const char *sslFingerprint = "2e cd e9 ee ff 45 5c 61 b4 a9 ca 53 47 39 d0 6e 97 46 51 7e";

// your network SSID (name) & network password
char myssid[] = " ";
char mypass[] = " ";

// unwiredlabs Hostname & Geolocation Endpoint url
const char* Host = "ap1.unwiredlabs.com";
String endpoint = "/v2/process.php";

// UnwiredLabs API_Token. Signup here to get a free token https://unwiredlabs.com/trial
String token = " "; 
String jsonString = "{\n";
// Variables to store unwiredlabs response
double latitude = 0.0;
double longitude = 0.0;
double accuracy = 0.0;

void setup() {
  Serial.begin(115200);

  // Set WiFi to station mode and disconnect from an AP if it was previously connected
  WiFi.mode(WIFI_STA);
  WiFi.disconnect();
  Serial.println("Setup done");

  // We start by connecting to a WiFi network
  Serial.print("Connecting to ");
  Serial.println(myssid);
  WiFi.begin(myssid, mypass);
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println(".");
}
void loop() {
  char bssid[6];
  DynamicJsonDocument doc(1024); //V6
  // WiFi.scanNetworks will return the number of networks found
    int n = WiFi.scanNetworks();
    Serial.println("scan done");
    if (n == 0 ) {
        Serial.println("No networks available");
    } else {
        Serial.print(n);
        Serial.println(" networks found");
    }
      // now build the jsonString...
      jsonString = "{\n";
      jsonString += "\"token\" : \"";
      jsonString += token;
      jsonString += "\",\n";
      jsonString += "\"id\" : \"saikirandevice01\",\n";
      jsonString += "\"wifi\": [\n";
      for (int j = 0; j < n; ++j) {
          jsonString += "{\n";
          jsonString += "\"bssid\" : \"";
          jsonString += (WiFi.BSSIDstr(j));
          jsonString += "\",\n";
          jsonString += "\"signal\": ";
          jsonString += WiFi.RSSI(j);
          jsonString += "\n";
          if (j < n - 1) {
              jsonString += "},\n";
          } else {
              jsonString += "}\n";
          }
      }
      jsonString += ("]\n");
      jsonString += ("}\n");
      Serial.println(jsonString);
      WiFiClientSecure client;
      //Connect to the client and make the api call
      Serial.println("Requesting URL: https://" + (String)Host + endpoint);
        client.setFingerprint(sslFingerprint);
    if (client.connect(Host, 443)) {
          Serial.println("Connected");
          client.println("POST " + endpoint + " HTTP/1.1");
          client.println("Host: " + (String)Host);
          client.println("Connection: close");
          client.println("Content-Type: application/json");
          client.println("User-Agent: Arduino/1.0");
          client.print("Content-Length: ");
          client.println(jsonString.length());
          client.println();
          client.print(jsonString);
          delay(500);
      }

      //Read and parse all the lines of the reply from server
      while (client.available()) {
          String line = client.readStringUntil('\r');
    auto error = deserializeJson(doc, line); //V6
    if (error)
    {
      Serial.print(F("deserializeJson() failed with code "));
      Serial.println(error.c_str());
      return;
    }
      latitude    = doc["lat"];
      longitude   = doc["lon"];
      accuracy    = doc["accuracy"];
      }
      Serial.println();
              Serial.print("Latitude = ");
              Serial.println(latitude, 6);
              Serial.print("Longitude = ");
              Serial.println(longitude, 6);
              Serial.print("Accuracy = ");
              Serial.println(accuracy);
      
      Serial.println("closing connection");
      Serial.println();
      client.stop();

      delay(5000);

}`

I got this error after running the code
`deserializeJson() failed with code Invalid input`

What am I doing wrong ?

bblanchon, 2021-06-17:
Hi @RohanMoj1999,

Please try the [ArduinoJson Troubleshooter](https://arduinojson.org/v6/troubleshooter/); it knows how to solve most issues.
Come back here if you need.

Best regards,
Benoit

closed 2021-07-03 by bblanchon