bblanchon/ArduinoJson #1591: Particle.io complier error with an empty program

sentientthings, 2021-06-17:
Compling the following with a Particle.io argon on deviceOS 2.1.0 (or 2.0.1) with ArduinoJson 6.18.0

```
#include "Particle.h"
#include <Arduino.h>
#include <ArduinoJson.h>

void setup() {

}

void loop() {
  // not used in this example
}
```

Gives the following raw compiler error:

```
attempting to compile firmware
Compile failed: Processing  lib/ArduinoJson/examples/JsonConfigFile/JsonConfigFile.ino
Processing  lib/ArduinoJson/examples/JsonFilterExample/JsonFilterExample.ino
Processing  lib/ArduinoJson/examples/JsonGeneratorExample/JsonGeneratorExample.ino
Processing  lib/ArduinoJson/examples/JsonHttpClient/JsonHttpClient.ino
Processing  lib/ArduinoJson/examples/JsonParserExample/JsonParserExample.ino
Processing  lib/ArduinoJson/examples/JsonServer/JsonServer.ino
Processing  lib/ArduinoJson/examples/JsonUdpBeacon/JsonUdpBeacon.ino
Processing  lib/ArduinoJson/examples/MsgPackParser/MsgPackParser.ino
Processing  lib/ArduinoJson/examples/ProgmemExample/ProgmemExample.ino
Processing  lib/ArduinoJson/examples/StringExample/StringExample.ino
Processing  src/ArduinoJsonTest.ino
Checking library ArduinoJson...
Installing library ArduinoJson 6.18.0 to lib/ArduinoJson ...
Library ArduinoJson 6.18.0 installed.
make -C ../modules/argon/user-part all
make[1]: Entering directory '/firmware/modules/argon/user-part'

mkdir -p target/
make -f ../../../modules/shared/nRF52840/build_linker_script.mk PREBUILD=1
make[2]: Entering directory '/firmware/modules/argon/user-part'
Creating target/platform_user_ram.ld ...
make[2]: Leaving directory '/firmware/modules/argon/user-part'

make -C ../../../user 
make[2]: Entering directory '/firmware/user'
Building cpp file: src/ArduinoJsonTest.cpp
Invoking: ARM GCC CPP Compiler
mkdir -p ../build/target/user/platform-12-msrc/
arm-none-eabi-gcc -DSTM32_DEVICE -DnRF52840 -DNRF52840_XXAA -DPLATFORM_THREADING=1 -DPLATFORM_ID=12 -DPLATFORM_NAME=argon -DPLATFORM_GEN=3 -DUSBD_VID_SPARK=0x2B04 -DUSBD_PID_DFU=0xD00C -DUSBD_PID_CDC=0xC00C -DSPARK_PLATFORM -g3 -gdwarf-2 -Os -mcpu=cortex-m4 -mthumb -mabi=aapcs -mfloat-abi=hard -mfpu=fpv4-sp-d16 --specs=nano.specs -DINCLUDE_PLATFORM=1 -DSOFTDEVICE_PRESENT=1 -DS140 -DPRODUCT_ID=12 -DPRODUCT_FIRMWARE_VERSION=65535 -D_WIZCHIP_=W5500 -DUSE_STDPERIPH_DRIVER -DDFU_BUILD_ENABLE -DLFS_CONFIG=lfs_config.h -DSYSTEM_VERSION_STRING=2.1.0 -DRELEASE_BUILD -I./inc -I../wiring/inc -I../system/inc -I../services/inc -I../third_party/nanopb/nanopb -I../communication/inc -I../hal/inc -I../hal/shared -I../hal/src/argon -I../hal/network/lwip/esp32 -I../hal/src/nRF52840 -I../hal/src/armv7 -I../hal/src/nRF52840 -I../hal/src/nRF52840/lwip -I../hal/src/nRF52840/freertos -I../hal/src/nRF52840/mbedtls -I../hal/src/nRF52840/littlefs -I../hal -I../hal/src/nRF52840/posix -I../hal/network -I../hal/network/api -I../hal/network/lwip -I../hal/network/lwip/posix -I../hal/network/lwip/wiznet -I../hal/network/ncp -I../hal/network/ncp/at_parser -I../hal/network/util -I../third_party/lwip/lwip/src/include -I../third_party/freertos/freertos/FreeRTOS/Source/include -I../third_party/nrf5_sdk/nrf5_sdk/external/freertos/portable/GCC/nrf52 -I../third_party/nrf5_sdk/nrf5_sdk/external/freertos/portable/CMSIS/nrf52 -I../third_party/wiznet_driver/wiznet_driver/Ethernet -I../gsm0710muxer/gsm0710muxer/include -I../platform/shared/inc -I../third_party/nrf5_sdk -I../third_party/nrf5_sdk/nrf5_sdk/components/toolchain/cmsis/include -I../third_party/nrf5_sdk/nrf5_sdk/modules/nrfx -I../third_party/nrf5_sdk/nrf5_sdk/modules/nrfx/drivers -I../third_party/nrf5_sdk/nrf5_sdk/modules/nrfx/drivers/include -I../third_party/nrf5_sdk/nrf5_sdk/modules/nrfx/hal -I../third_party/nrf5_sdk/nrf5_sdk/modules/nrfx/mdk -I../third_party/nrf5_sdk/nrf5_sdk/modules/nrfx/hal -I../third_party/nrf5_sdk/nrf5_sdk/modules/nrfx/drivers/src/prs -I../third_party/nrf5_sdk/nrf5_sdk/components/libraries/util -I../third_party/nrf5_sdk/nrf5_sdk/components/softdevice/s140/headers -I../third_party/nrf5_sdk/nrf5_sdk/components/softdevice/s140/headers/nrf52 -I../third_party/nrf5_sdk/nrf5_sdk/components/softdevice/common -I../third_party/nrf5_sdk/nrf5_sdk/components/softdevice/mbr/nrf52840/headers -I../third_party/nrf5_sdk/nrf5_sdk/components/libraries/fstorage -I../third_party/nrf5_sdk/nrf5_sdk/components/libraries/experimental_section_vars -I../third_party/nrf5_sdk/nrf5_sdk/components/libraries/atomic -I../third_party/nrf5_sdk/nrf5_sdk/components/libraries/crc32 -I../third_party/nrf5_sdk/nrf5_sdk/integration/nrfx/legacy -I../third_party/nrf5_sdk/nrf5_sdk/components/libraries/delay -I../third_party/nrf5_sdk/nrf5_sdk/components/libraries/log -I../third_party/nrf5_sdk/nrf5_sdk/components/libraries/log -I../third_party/nrf5_sdk/nrf5_sdk/components/libraries/log/src -I../third_party/nrf5_sdk/nrf5_sdk/components/libraries/experimental_section_vars -I../third_party/nrf5_sdk/nrf5_sdk/components/libraries/atomic -I../third_party/nrf5_sdk/nrf5_sdk/components/libraries/delay -I../third_party/nrf5_sdk/nrf5_sdk/components/libraries/queue -I../third_party/nrf5_sdk/nrf5_sdk/components/softdevice/common -I../third_party/nrf5_sdk/nrf5_sdk/components/softdevice/s140 -I../third_party/nrf5_sdk/nrf5_sdk/components/softdevice/s140/headers -I../third_party/nrf5_sdk/nrf5_sdk/components/softdevice/s140/headers/nrf52 -I../third_party/nrf5_sdk/nrf5_sdk/components/libraries/fifo -I../third_party/nrf5_sdk/nrf5_sdk/components/libraries/atomic_fifo -I../third_party/nrf5_sdk/nrf5_sdk/components/libraries/strerror -I../third_party/nrf5_sdk/nrf5_sdk/components/libraries/fstorage -I../third_party/nrf5_sdk/nrf5_sdk/components/libraries/atomic_flags -I../third_party/nrf5_sdk/nrf5_sdk/components/libraries/usbd -I../third_party/nrf5_sdk/nrf5_sdk/components/libraries/usbd/class/cdc -I../third_party/nrf5_sdk/nrf5_sdk/components/libraries/usbd/class/cdc/acm -I../third_party/nrf5_sdk/nrf5_sdk/components/ble/common -I../third_party/nrf5_sdk/nrf5_sdk/components/ble/nrf_ble_gatt -I../third_party/nrf5_sdk/nrf5_sdk/components/ble/ble_advertising -I../third_party/nrf5_sdk/nrf5_sdk/components/nfc/t2t_lib -I../third_party/nrf5_sdk/nrf5_sdk/components/nfc/platform -I../third_party/nrf5_sdk/nrf5_sdk/components/nfc/ndef/generic/message -I../third_party/nrf5_sdk/nrf5_sdk/components/nfc/ndef/generic/record -I../third_party/nrf5_sdk/nrf5_sdk/components/nfc/ndef/lauchapp -I../third_party/nrf5_sdk/nrf5_sdk/components/nfc/ndef/text -I../third_party/nrf5_sdk/nrf5_sdk/components/nfc/ndef/uri -I../third_party/nrf5_sdk/nrf5_sdk/external/nrf_cc310/include -I../third_party/nrf5_sdk/nrf5_sdk/external/utf_converter -I../third_party/littlefs/littlefs -I../third_party/miniz/miniz -I../platform/MCU/nRF52840/inc -I../dynalib/inc -Isrc -I./libraries -Ilib/ArduinoJson/src -I. -MD -MP -MF ../build/target/user/platform-12-msrc/ArduinoJsonTest.o.d -ffunction-sections -fdata-sections -Wall -Wno-switch -Wno-error=deprecated-declarations -fmessage-length=0 -fno-strict-aliasing -DSPARK=1 -DPARTICLE=1 -Wundef -DSTART_DFU_FLASHER_SERIAL_SPEED=14400 -DSTART_YMODEM_FLASHER_SERIAL_SPEED=28800 -DSPARK_PLATFORM_NET=ESP32 -fno-builtin-malloc -fno-builtin-free -fno-builtin-realloc  -DLOG_INCLUDE_SOURCE_INFO=1 -DPARTICLE_USER_MODULE -DUSER_FIRMWARE_IMAGE_SIZE=0x20000 -DUSER_FIRMWARE_IMAGE_LOCATION=0xD4000 -DMODULAR_FIRMWARE=1 -DMODULE_VERSION=6 -DMODULE_FUNCTION=5 -DMODULE_INDEX=1 -DMODULE_DEPENDENCY=4,1,2101 -DMODULE_DEPENDENCY2=0,0,0 -D_GNU_SOURCE -DLOG_MODULE_CATEGORY="\"app\""  -fno-exceptions -fno-rtti -fcheck-new -flto -ffat-lto-objects -DPARTICLE_COMPILE_LTO_FAT -std=gnu++14 -c -o ../build/target/user/platform-12-msrc/ArduinoJsonTest.o src/ArduinoJsonTest.cpp
In file included from ./inc/Arduino.h:26,
                 from src/ArduinoJsonTest.cpp:10:
lib/ArduinoJson/src/ArduinoJson/Polyfills/pgmspace_generic.hpp: In instantiation of 'typename ArduinoJson6180_91::enable_if<ArduinoJson6180_91::is_pointer<T>::value, T>::type ArduinoJson6180_91::pgm_read(const void*) [with T = const __FlashStringHelper*; typename ArduinoJson6180_91::enable_if<ArduinoJson6180_91::is_pointer<T>::value, T>::type = const __FlashStringHelper*]':
lib/ArduinoJson/src/ArduinoJson/Deserialization/DeserializationError.hpp:100:12:   required from here
../wiring/inc/avr/pgmspace.h:148:29: error: 'const void*' is not a pointer-to-object type
  148 | #define pgm_read_ptr(addr) (*(const void *)(addr))
      |                            ~^~~~~~~~~~~~~~~~~~~~~~
../build/module.mk:274: recipe for target '../build/target/user/platform-12-msrc/ArduinoJsonTest.o' failed
[... cut ...]
make[1]: Leaving directory '/firmware/modules/argon/user-part'
../build/recurse.mk:11: recipe for target 'modules/argon/user-part' failed
make: *** [modules/argon/user-part] Error 2
	at makeError (c:\Users\rober\.vscode\extensions\particle.particle-vscode-core-1.13.8\node_modules\execa\lib\error.js:59:11)
	at handlePromise (c:\Users\rober\.vscode\extensions\particle.particle-vscode-core-1.13.8\node_modules\execa\index.js:114:26)
	at runMicrotasks (<anonymous>)
	at processTicksAndRejections (internal/process/task_queues.js:97:5)

- - - -
```

bblanchon, 2021-06-18:
Hi,

We saw the error `'const void*' is not a pointer-to-object type` with several boards already:

* #1381, with Teknic ClearCore
* #1442, with ESP8266
* #1519, with Industruino
* #1536, but the user didn't tell which board it was

In each case, this comes from a bug in the [Arduino Core API](https://github.com/arduino/ArduinoCore-API).
This bug was fixed by arduino/ArduinoCore-API#118 in October 2020, but wasn't immediately propagated to all derivatives.

First, make sure that every related piece of software is up-to-date.
If the problem persists, please open an issue in the relevant project and point them to arduino/ArduinoCore-API#118.

As a workaround, you can disable support for `PROGMEM` in ArduinoJson. See [ARDUINOJSON_ENABLE_PROGMEM](https://arduinojson.org/v6/api/config/enable_progmem/).

Best regards,
Benoit

sentientthings, 2021-06-18:
"As a workaround, you can disable support for PROGMEM in ArduinoJson. See ARDUINOJSON_ENABLE_PROGMEM."

Worked.  Thank you.

closed 2021-06-18 by sentientthings