bblanchon/ArduinoJson #1592: `build-single-header.sh` fails

jjsuwa-sys3175, 2021-06-19:
gcc version 8.3.0 (Raspbian 8.3.0-6+rpi1) :
```
foo@debootstrap:/tmp/ArduinoJson$ extras/scripts/build-single-header.sh 
src/ArduinoJson.h -> ArduinoJson.hpp
   (*snip*)
ArduinoJson/MsgPack/MsgPackSerializer.hpp -> ArduinoJson/Variant/VariantData.hpp
ArduinoJson.hpp -> ArduinoJson/compatibility.hpp
In file included from <stdin>:1:
../ArduinoJson-5ccd62a.h:697:12: fatal error: ArduinoJson/Strings/StdStringAdapter.hpp: No such file or directory
 #  include <ArduinoJson/Strings/StdStringAdapter.hpp>
            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
```

bblanchon, 2021-06-19:
Hi @jjsuwa-sys3175,

Thank you very much for reporting this bug.
It's about time I add this to the CI anyway.

Best regards,
Benoit

closed 2021-06-21 by bblanchon

bblanchon, 2021-06-21:
You can now download the single header distribution from the build artifacts.

1. [Open CI builds history](https://github.com/bblanchon/ArduinoJson/actions/workflows/ci.yml)
2. Click on the relevant run (for example, the latest in branch `6.x`)
3. Scroll at the bottom to download the "Single headers" artifact.

bblanchon, 2021-06-21:
[image]

commit 8e1ac0c8 (2021-06-21): Fixed build-single-header.sh (fixes #1592)
commit 64f56f9a (2021-06-21): Fixed build-single-header.sh (fixes #1592)
commit 9182c206 (2021-06-21): Fixed build-single-header.sh (fixes #1592)