bblanchon/ArduinoJson #1593: Slow Deserialization (even when using ReadBufferingStream)

swabygw, 2021-06-21:
The VS Output window shows a read about one time per second when the code is this:

	DynamicJsonDocument doc(1024);
	ReadBufferingStream bufferedStream(Serial, 100);
	DeserializationError err = deserializeJson(doc, bufferedStream);

[image]

The VS Output window shows a read about four (or five) times per second when the code is this:

	DynamicJsonDocument doc(1024);
	char json[] = "{\"sensor\":\"gps\",\"time\":1351824120,\"data\":[48.756080,2.302038]}";
	DeserializationError err = deserializeJson(doc, json);

[image]

But I need to deserialize from Serial.  Is there another way to speed up deserialization from Serial?

bblanchon, 2021-06-21:
Hi @swabygw,

I ran the same tests on my Leonardo (which has the same performance as the UNO) and got the following results:

* Deserialize from `char[]`: **640 µs**
* Deserialize from `Serial`: **1280 µs**

These are microseconds, not milliseconds!

I think that your measurement method is wrong: you probably include the time to way for the input.
[Here are the files I used for my measurementd](https://gist.github.com/bblanchon/1ddbd7f37a254ba2378c3beb5af535b5).

`Serial` is a buffered stream, so `ReadBufferingStream` cannot improve performance (**2050 µs** from my experiment).
Please let me know if you found some contradictory information in the documentation.

BTW, I recommend using a [`StaticJsonDocument`](https://arduinojson.org/v6/api/staticjsondocument/) because the UNO's RAM is too small for a heap, and it's significantly faster.

Best regards,
Benoit

swabygw, 2021-06-21:
Thanks - I was missing the "if (!Serial.available()) return;" - that did it.

I noticed something else, too, though, that if I tried to show the incoming data in the terminal, it paused the program.  I show the incoming data in the terminal like this:

     Serial.print("data received: "); serializeJson(doc, Serial); Serial.println();

And it pauses the program.  The VS Output window will occasionally show a message that I should use a "println" function to terminate the printing.  Perhaps there is a better way to display the incoming data string?

bblanchon, 2021-06-22:
I don't see any reason why `serializeJson(doc, Serial);`.
However, `deserializeJson(doc, Serial)` could pause the execution if the input is empty.

This happened to me while running the test because I initially had the Serial Monitor configured to send CR+LF after the message.
So, when I sent `{"sensor":"gps","time":1351824120,"data":[48.756080,2.302038]}`, what the Arduino really received was `{"sensor":"gps","time":1351824120,"data":[48.756080,2.302038]}\r\n`.
Since `deserializeJson()` stops reading immediately at the end of the object, the `\r\n` remains in the serial buffer.
Therefore, `Serial.available()` returns `2` and the program calls `deserializeJson()` again.
`deserializeJson()` reads `\r\n`, which are just white spaces for him, so it continues reading until a timeout occurs (1s by default), and it finally returns `EmptyInput`.

If you cannot remove the CR+LF at the end of the message, you must add a flush loop after `deserializeJson(doc, Serial)`, like so:

```c++
while (isspace(Serial.peek()) Serial.read();
```

swabygw, 2021-06-22:
I updated the deserialization line to the following:
	DeserializationError err = deserializeJson(doc, Serial);
	while (isspace(Serial.peek())) { Serial.read(); }

But it didn't solve the pausing.  I tried sending data both with  
[image]
 and 
[image]
.  Still had the problem.  But, actually, I really still think it's the serializationJson because just removing the serializationJson from this line:

  Serial.print("data received: "); **serializeJson(doc, Serial);** Serial.println();

...solves the problem.

bblanchon, 2021-06-23:
Again, I don't see any reason why `serializeJson(doc, Serial)` would freeze the program.
Alternatively, you can serialize to a `String` or a `char[]` and print the string.
For example:

```c++
char json[256];
size_t n = serializeJson(doc, json);
Serial.write(json, n);
```

swabygw, 2021-06-23:
I can't figure out what's going on, but deserializing and serializing don't play well together.  I think that the serialization is too slow and the loop is running faster.  Sometimes the serialization prints out some of the statements, sometimes all of the statements, sometimes parts of the statements, etc.  A few times I've even got a message of "NoMemory" from the deserialization.  I'm printing out a copy of the program below (it's clearly used to run a motor controller).  Please give it a try and let me know if you see some improvements that could be made.

	#include <ArduinoJson.hpp>
	#include <ArduinoJson.h>
	#include <StreamUtils.h>

	unsigned long myTime;
	const int MOTOR_LEFT = 5; //analog pin 5
	const int MOTOR_RIGHT = 6; //analog pin 6

	// the setup function runs once when you press reset or power the board
	void setup() {

		// initialize pins
		pinMode(LED_BUILTIN, OUTPUT);
		pinMode(MOTOR_LEFT, OUTPUT);
		pinMode(MOTOR_RIGHT, OUTPUT);

		// Initialize serial port
		Serial.begin(115200);
		while (!Serial) continue;

	}

	// the loop function runs over and over again forever
	void loop() {

	  //store milliseconds since startup
	  myTime = millis();  

	  //ReadAnalogInputs();
	  //Serialization();  
	  Deserialization();    

	}

	String timestamp() {

		char out[10];
		sprintf(out, "(%i) ", myTime);
		return out;
	}

	void Deserialization() {

		StaticJsonDocument<1024> doc;

		if (!Serial.available()) return;

		DeserializationError err = deserializeJson(doc, Serial);
		while (isspace(Serial.peek())) {
			Serial.read();
		}

		Serial.println();
		int i; for (i = 0; i < 50; i++) { Serial.print("="); }; Serial.println();
		Serial.print(timestamp());
		Serial.print(F("deserializeJson() completed with code: "));
		Serial.println(err.c_str());

		if (err.c_str() == "Ok") {
			Serial.print(timestamp());
			Serial.print("data received: "); serializeJson(doc, Serial); Serial.println();
			ParseData(myTime, doc);
		}
		else if (err.c_str() == "EmptyInput") {
			Serial.print(timestamp());
			Serial.println(F("deserializeJson() contained no data"));
		}
		else {
			Serial.print(timestamp());
			Serial.print(F("deserializeJson() failed with code: ")); Serial.println(err.c_str());
		}

		return;

	}

	void ParseData(unsigned long myTime, StaticJsonDocument<200> doc) {

		Serial.print(timestamp());
		Serial.print("parsed data: "); serializeJson(doc, Serial); Serial.println();

		const char* sensor = doc["sensor"];
		long time = doc["time"];
		double latitude = doc["data"][0];
		double longitude = doc["data"][1];

		double midpoint = 128.0;
		int voltage_min = 0;
		int voltage_max = 255;

		String motor = doc["motor"] | "Error"; // valid values: "Error", "Left", "Right"
		double speed = doc["speed"] | 0.0; // valid values: -1.00 to 1.00, 0.00 is center
		int volts = ((speed * midpoint) + midpoint); // valid values: 0 to 255, yields 0v to 5v (5v logic)
		
		String out = timestamp();
		out.concat("motor: "); out.concat(motor);
		out.concat(", speed: "); out.concat(speed);
		out.concat(", volts: "); out.concat(volts);
		
		Serial.println(out);  // sometimes this prints out, sometimes not

		return;  // stop here

		if (motor == "Error") {
			Serial.print(timestamp()); Serial.print("Invalid motor value: "); Serial.println(motor);
		}
		if ((volts < 0) || (volts > 255)) {
			Serial.print(timestamp()); Serial.print("Invalid volts calculation: "); Serial.println(volts);
		}

		if (motor == "Left") { analogWrite(MOTOR_LEFT, min(voltage_max, max(voltage_min, volts))); }
		if (motor == "Right") { analogWrite(MOTOR_RIGHT, min(voltage_max, max(voltage_min, volts))); }

	}

swabygw, 2021-06-23:
I rewrote the loop without calling functions and the following works perfectly:

	void loop() {

	  //store milliseconds since startup
	  myTime = millis();  

	  //ReadAnalogInputs();
	  //Serialization();  
	  //Deserialization();  

	  String inData;
	  StaticJsonDocument<1024> doc;
	  while (Serial.available() > 0)
	  {
		  inData += Serial.readStringUntil('}');

		  DeserializationError err = deserializeJson(doc, inData);
		  Serial.print(timestamp());
		  Serial.print("data received: "); serializeJson(doc, Serial); Serial.println();

		  const char* sensor = doc["sensor"];
		  long time = doc["time"];
		  double latitude = doc["data"][0];
		  double longitude = doc["data"][1];

		  double midpoint = 128.0;
		  int voltage_min = 0;
		  int voltage_max = 255;

		  String motor = doc["motor"] | "Error"; // valid values: "Error", "Left", "Right"
		  double speed = doc["speed"] | 0.0; // valid values: -1.00 to 1.00, 0.00 is center
		  int volts = ((speed * midpoint) + midpoint); // valid values: 0 to 255, yields 0v to 5v (5v logic)

		  String out = timestamp();
		  out.concat("motor: "); out.concat(motor);
		  out.concat(", speed: "); out.concat(speed);
		  out.concat(", volts: "); out.concat(volts);
		  Serial.println(out);

		  if (motor == "Error") {
			  Serial.print(timestamp()); Serial.print("Invalid motor value: "); Serial.println(motor);
		  }
		  if ((volts < 0) || (volts > 255)) {
			  Serial.print(timestamp()); Serial.print("Invalid volts calculation: "); Serial.println(volts);
		  }

		  if (motor == "Left") { analogWrite(MOTOR_LEFT, min(voltage_max, max(voltage_min, volts))); }
		  if (motor == "Right") { analogWrite(MOTOR_RIGHT, min(voltage_max, max(voltage_min, volts))); }

		  return; // stop here

	  }

	}

This leads me to believe that perhaps something was wrong with my constructor for the ParseData function when it created a new StaticJsonDocument.

swabygw, 2021-06-23:
Updating the constructor to...

	void ParseData(StaticJsonDocument<1024>& doc)

...fixed everything.

bblanchon, 2021-06-24:
I'm very glad you figure out how to solve the issue!

I would just change the signature of `ParseData` to:

```c++
void ParseData(JsonDocument& doc)
```

This way, you can use any type of `JsonDocument`.

swabygw, 2021-06-24:
Done - thanks!

closed 2021-06-25 by bblanchon