bblanchon/ArduinoJson #1594: desserializeErro NoMemory

filipedias1994, 2021-06-21:
Hi, I'm using an ESP32 and getting the following json from mqtt, but when deserializing I get the error NoMemory and capacity() = 0, I saw in the documentation that this is a RAM memory problem (https://arduinojson.org/v6/how-to/reduce-memory-usage/), but I can't do any of the tips offered.

I try to filter but I can't filter this json as described on the website. Decreases the amount of String but I get the same thing, NoMemory and capacity() = 0.

How could I do to reduce this json? 
How can I filter this json only with the fields "document", "keycode", for example?

JSON:
`{"41201":{"bookingNumber":"","number":41201,"shippedDateStr":"09/06/2021 - 15:37:24","dataStatus":"Porta fechada","sensorPin":18,"active":true,"lockerSize":1,"createdAt":{"_seconds":1623695942,"_nanoseconds":407000000},"updatedAt":{"_seconds":1623695928,"_nanoseconds":746000000},"keycode":"489182","document":"12345678910","status":"Sync","bookedDate":{"_seconds":1623695942,"_nanoseconds":957000000},"deliveredDateStr":"09/06/2021 - 15:39:28","syncDateStr":"21/06/2021 - 19:44:46","nrOfDays":"2","relayPin":23,"qrCode":"10110000000000960003813456212186620210614154122ae3e5f85775b9b0c027f22e403b3a28e"}}`

CODE:
`void criarClasse(const char* json)
{
  DynamicJsonDocument doc(30000);
  /*StaticJsonDocument<60000> filter;
  filter[0]["document"] = true;
  filter[0]["keycode"] = true;
  filter[0]["number"] = true;
  filter[0]["status"] = true;
  filter[0]["sensorPin"] = true;
  filter[0]["relayPin"] = true;
  filter[0]["nrOfDays"] = true;
  filter[0]["bookingNumber"] = true;
  filter[0]["dataStatus"] = true;*/
  DeserializationError err = deserializeJson(doc, json /*DeserializationOption::Filter(filter)*/);
  if (err) {
    dbSerialPrintln(F("deserializeJson() failed: "));
    dbSerialPrintln(err.c_str());
  }
  dbSerialPrintln("Capacidade: ");
  dbSerialPrintln(doc.capacity());
 
  JsonObject root = doc.as<JsonObject>();
  
  int count = -1;
  String pub = "esp32/xxxr/" + receivedData;
  for (JsonPair kv : root)
  {
    count++;
    test[count].id = kv.key().c_str();
    test[count].document = (const char*)doc[lockers[count].id]["document"];
    test[count].keycode = (const char*)doc[lockers[count].id]["keycode"];
    test[count].number = doc[lockers[count].id]["number"];
    test[count].status = (const char*)doc[lockers[count].id]["status"];
    test[count].sensorPin = doc[lockers[count].id]["sensorPin"];
    test[count].relayPin =  doc[lockers[count].id]["relayPin"];
    test[count].nrOfDays = (const char*)doc[lockers[count].id]["nrOfDays"];
    test[count].bookingNumber = (const char*)doc[lockers[count].id]["bookingNumber"];
    test[count].dataStatus = (const char*)doc[lockers[count].id]["dataStatus"];
  }
}`

bblanchon, 2021-06-22:
Hi,

The filter for "document" and "keycode":

```json
{
  "*": {
    "document": true,
    "keycode": true
  }
}
```

Use the ArduinoJson Assistant to compute the required capacity.

If this still isn't working, you should try to deserialize in chunks as described in [How to reduce memory usage?](https://arduinojson.org/v6/how-to/reduce-memory-usage).
I'm surprised you manage to transmit such large messages over MQTT; what library do you use?

Best regards,
Benoit

filipedias1994, 2021-06-22:
Thanks for the answer!

I can't understand how to perform this filter, the example in the documentation leaves me with doubts. For example, a StaticJsonDocument is used, but could you use it as Dynamic? Another question is that the "doc" variable is not declared on the site. I don't need to declare myself?

StaticJsonDocument<64> filter;
filter["list"][0]["temperature"] = true;
deserializeJson(doc, input, DeserializationOption::Filter(filter));

I use PubSub library to publish and receive MQTT messages.

I used Arduino Assistant, it reports about 25000 bytes, I left it with 30000 and it still doesn't work. It successfully deserializes the first few times, but if I get this same JSON with some values ​​from the modified fields, it gives NoMemory error with capacity 0.

Thanks

bblanchon, 2021-06-23:
Your main issue is the capacity of 0:  it means that the `DynamicJsonDocument` fails to allocate the memory pool.
I understand that the first time it works and then it starts failing, which means that either:

* there is a memory leak in your program, or
* there is too much [heap fragmentation](https://cpp4arduino.com/2018/11/06/what-is-heap-fragmentation.html)

Now, it's up to you to work on these issues; the only thing I can tell you for sure, it's that the problem doesn't come from ArduinoJson.

closed 2021-07-03 by bblanchon