bblanchon/ArduinoJson #1595: How to get parameters of DynamicJsonDocument created by new operator

sh-user, 2021-06-23:
I need to create a DynamicJsonDocument object with the new operator, 
read the parameters, and then clear the memory with the delete operator

Example of my code on ESP8266:
```
bool ir_remout;
uint8_t num;
char ssid[32];
char pass[32];

void ReadConfig() {
  File file = LittleFS.open("/json","r");
  if (file) {
    DynamicJsonDocument *doc = new DynamicJsonDocument(2500);
    DeserializationError err = deserializeJson(*doc, file);
    file.close();
    if (!err) {
       num = doc["num"].as<uint8_t>();
       ir_remout = doc["ir"].as<bool>();
       strlcpy(ssid, doc["ssid"] | "", sizeof(ssid));
       strlcpy(pass, doc["pass"] | "", sizeof(pass));
    }
    doc->clear();
    delete doc;
    doc = NULL;
  }
}
```
Error
`invalid types 'ArduinoJson6180_91::DynamicJsonDocument* {aka ArduinoJson6180_91::BasicJsonDocument<ArduinoJson6180_91::DefaultAllocator>*}[const char [4]]' for array subscript`

bblanchon, 2021-06-24:
Hi @sh-user,

Why would you do that?
I don't see any reason to use `new` with `DynamicJsonDocument`...

Anyway, if you absolutely want to do that, you must dereference the pointer, e.g. `(*doc)["ssid"]`.

Best regards,
Benoit

sh-user, 2021-06-24:
Thanks for the answer, `(*doc)` helped. But I did not notice the difference in memory. I need to save memory as much as possible.
 I think that DynamicJsonDocument allocates 2500 bytes and does not free memory, so I started looking for different ways to clear this memory. Tried using the base class destructor `doc.~BasicJsonDocument()`, but `ESP.getFreeHeap()` returns unrealistic 90K bytes. Perhaps somewhere is my mistake.

bblanchon, 2021-06-25:
What makes you think `DynamicJsonDocument` doesn't free memory?

sh-user, 2021-06-25:
Perhaps I mistakenly believe that heap allocated memory is not cleared. For example, stack memory is cleared automatically.

bblanchon, 2021-06-27:
If you use a `DynamicJsonDocument` as an [automatic variable](https://en.wikipedia.org/wiki/Automatic_variable) (which you should), it automatically deallocates the heap memory, just as `String` does.

closed 2021-07-03 by bblanchon