bblanchon/ArduinoJson #1596: Getting a string from filtered input on ESP32

jbellue, 2021-06-25:
Hi,
I'm trying to filter out a string from openweathermap on an ESP32, but it comes out as null.
If I don't filter the extra data, everything works fine.
Here's an example (the received JSON data is `[{"name":"Puynormand","local_names":{"ascii":"Puynormand","feature_name":"Puynormand"},"lat":44.9833,"lon":null,"country":"FR"}]`):
```
#include <Arduino.h>
#include <HTTPClient.h>
#include <ArduinoJson.h>
#include "WiFi.h"

void setup() {
  Serial.begin(115200);
  WiFi.begin();
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }

  HTTPClient http;
  http.useHTTP10(true);
  http.begin("https://api.openweathermap.org/geo/1.0/reverse?lat=45.0000&lon=0.0000&limit=1&appid=REDACTED");
  const int httpResponseStatus = http.GET();
  if (httpResponseStatus > 0) {
    StaticJsonDocument<256> filter;
    filter["cod"] = true;
    filter["message"] = true;
    filter[0]["name"] = true;

    StaticJsonDocument<256> doc;
    DeserializationError error = deserializeJson(doc, http.getStream(), DeserializationOption::Filter(filter));

    if (error) {
      Serial.println("deserialization error");
    }
    else {
      Serial.print(doc[0]["name"].as<String>());
    }
  }
  else {
    Serial.println(http.errorToString(httpResponseStatus).c_str());
  }
  http.end();
}

void loop() {
  // unused
}
```

The output of `doc[0]["name"]` is `null`.
When commenting out the two unused filters (`filter["cod"] = true;` and `filter["message"] = true;` the string gets displayed as it should.

The two unused filters are used to detect API issues (wrong API key for example).

I am using the latest version of platformio to compile and upload to the ESP32.

bblanchon, 2021-06-27:
Hi @jbellue,

The filter is incorrect. If you do a `serializeJsonPretty(filter, Serial)`, you'll see:

```json
{
  "cod": true,
  "message": true,
}
```

As you can see, the statement `filter[0]["name"] = true;` is ignored.
Indeed, `filter[0]` only works if `filter` contains an array, but in your case, it contains an object.

I'm not sure what you tried to achieve with `cod` and `message`, so I cannot tell you how to fix your program.
However, I highly recommend that you use the [ArduinoJson Assistant](https://arduinojson.org/v6/assistant/) to write the skeleton.

Best regards,
Benoit

jbellue, 2021-06-28:
Hi Benoît,

Sorry, I did forget to include the reason for the `cod` and `message` filters. If there's an API error, the returned JSON will include some error message: `{"cod":401, "message": "Invalid API key. Please see http://openweathermap.org/faq#error401 for more info."}`.

Using the assistant, I can't manage to parse both possible messages.

From what you're saying, it seems impossible to filter both a root array, and an arbitrary root key, is that correct?

Thanks

bblanchon, 2021-06-29:
Hi @jbellue,

You're right: ArduinoJson's filter doesn't support both at the same time, because the filter itself is a JSON document.
However, you can use a different filter based on the HTTP status code.

Best regards,
Benoit

Best regards,
Benoit

jbellue, 2021-06-30:
Hi,

That makes sense, thanks.
For some reason, I thought that the openweathermap API would always return a 200 HTTP status code, and you had to parse the `cod` message to figure out what went wrong.

Thanks a lot for your time!

closed 2021-06-30 by jbellue

bblanchon, 2021-07-01:
You're welcome, @jbellue!
Good luck with your project.