bblanchon/ArduinoJson #1597: ArduinoJson::JsonArray has no member named 'clear'

bhcuong2008, 2021-06-26:
Hi,

I create a JsonArray as below, then got error `'ArduinoJson::JsonArray {aka class ArduinoJson6180_D1::ArrayRef}' has no member named 'clear'`

My target platform: ESP32 Arduino 1.06, IDF 3.3.5
ArduinoJson 1.18

```
StaticJsonDocument<512> test;
JsonArray ja = test.as<JsonArray>();
ja.clear();
```
I dont know why because the document says it has method clear.

Thanks.

bblanchon, 2021-06-26:
Hi @bhcuong2008,

Thank you very much for reporting this bug 👍 
I fixed it in the main branch.

Best regards,
Benoit

closed 2021-06-26 by bblanchon

bhcuong2008, 2021-06-26:
Thank u very much for your fast fix.

bblanchon, 2021-07-03:
The fix is included in [ArduinoJson 6.18.1](https://github.com/bblanchon/ArduinoJson/releases/tag/v6.18.1)

commit a6fc9311 (2021-06-26): Added JsonArray::clear() (fixes #1597)