bblanchon/ArduinoJson #1598: Feature request: filter a number of occurence on a list

sn0oz, 2021-06-26:
Would it be possible to filter a number of occurence on a list to reduce further the data size for APIs which don't support pagination ?

For example this JSON:
```
{
list: [
    { "number": 0 },
    { "number": 1 },
    { "number": 2 },
    { "number": 3 }
    ]
}
```
filtered with something like:
`filter["list"][2]["number"] = true;`

would result in:
```
{
list: [
    { "number": 0 },
    { "number": 1 }
    ]
}
```

bblanchon, 2021-06-27:
Hi @sn0oz,

This feature was requested several times, but I never implemented it because I could not find a satisfying API.
See [this older answer](https://github.com/bblanchon/ArduinoJson/issues/1316#issuecomment-660510456) for details.

The API you suggest makes sense, but it's too limited and is inconsistent with the existing `filter["list"][0]["number"] = true`, which includes all the elements.  
Moreover, because ArduinoJson uses a dense array (as opposed to sparsed arrays), if a user sets `filter["list"][100]["number"] = true`, it allocates an array of 100 elements, which is quite a waste of RAM.

A better solution would be to use a query language as I suggested in https://github.com/bblanchon/ArduinoJson/issues/1486#issuecomment-778051459.
I'm not against the idea, but I'm afraid this will be a huge work with very little return, and the executable would inflate significantly.

We could also use a predicate instead of a query or filter.
This could be as powerful as the query language but wouldn't require a new parser and keep the code small.
I had this idea on the back of my head for a long time, but I couldn't figure out an intuitive API.

For your particular issue, I think the ["deserialization in chunks" technique](https://arduinojson.org/v6/how-to/deserialize-a-very-large-document/#deserialization-in-chunks) provides a better solution.

Best regards,
Benoit

sn0oz, 2021-06-27:
Hi Benoit, thanks for the detailed reply.
I understand your points, and will look at the deserialization in chuncks technique.
Thanks again for your great work on this library !

closed 2021-06-27 by sn0oz