bblanchon/ArduinoJson #1600: Update a link (error 404) in documentation

fabianoriccardi, 2021-06-27:
Hi Benoit,
The link to the benchmark of MessagePack vs JSON (#1211) is recently changed. It was linked at the end of [this page](https://arduinojson.org/v6/api/msgpack/serializemsgpack/) and now the link is 

    https://gist.github.com/fabianoriccardi/40bf4247f4d762dd483a765c0c2cba41

 In the meaning while I've updated all the benchmarks,  re-running them on the latest versions of Arduino Core and your ArduinoJson library.

bblanchon, 2021-06-27:
Fixed!

closed 2021-06-27 by bblanchon

fabianoriccardi, 2021-07-10:
Hi, I have updated and heavily refactored the benchmark, adding a Jupyter Notebook to generate charts with Plotly. So I had created a full git repository. 

https://github.com/fabianoriccardi/benchmark-json-messagepack

If your have suggestions they are welcomed :)

bblanchon, 2021-07-10:
Link updated!