bblanchon/ArduinoJson #1601: Storing positive value as an int can't be read as unsigned int

embedded-creations, 2021-06-29:
If I store a positive signed value (e.g. `top["testInt"] = (int)43;`) I can't read the same value out into an unsigned variable (`unsigned int tempUInt = top["testInt"];`), I just get 0.

This is probably an odd use case but I do have a good reason for doing this in some software where user extendable code is communicating with the main library code only through an ArduinoJson document.

This simple example shows the problem:

```
#include <ArduinoJson.h>

void setup() {
  Serial.begin(9600);
  while (!Serial);
  delay(1000);

  StaticJsonDocument<32> top;

  top["testInt"] = (int)43;
  int tempInt = top["testInt"];
  unsigned int tempUInt = top["testInt"];
  //top["testUInt"] = (unsigned int)43;

  Serial.print("tempInt = top[\"testInt\"]: ");
  Serial.println(tempInt);
  Serial.print("tempUInt = top[\"testInt\"]: ");
  Serial.println(tempUInt);

  Serial.print("top[\"testInt\"].is<int>(): ");
  Serial.println(top["testInt"].is<int>());
  Serial.print("top[\"testInt\"].is<unsigned int>(): ");
  Serial.println(top["testInt"].is<unsigned int>());
}

void loop() {

}
```
Result:
```
tempInt = top["testInt"]: 43
tempUInt = top["testInt"]: 0
top["testInt"].is<int>(): 1
top["testInt"].is<unsigned int>(): 0
```

Tested on Teensy 3.6 and ESP32.  Compiler output from Teensy 3.6:

```
/Applications/Teensyduino.app/Contents/Java/arduino-builder -dump-prefs -logger=machine -hardware /Applications/Teensyduino.app/Contents/Java/hardware -hardware /Users/louis/Library/Arduino15/packages -tools /Applications/Teensyduino.app/Contents/Java/tools-builder -tools /Applications/Teensyduino.app/Contents/Java/hardware/tools/avr -tools /Users/louis/Library/Arduino15/packages -built-in-libraries /Applications/Teensyduino.app/Contents/Java/libraries -libraries /Users/louis/Dropbox/Dev/Arduino/libraries -fqbn=teensy:avr:teensy36:usb=serial,speed=180,opt=o2std,keys=en-us -ide-version=10813 -build-path /var/folders/n6/l3jqlyms0r37qb1s_swjcqf00000gn/T/arduino_build_327691 -warnings=all -build-cache /var/folders/n6/l3jqlyms0r37qb1s_swjcqf00000gn/T/arduino_cache_65962 -verbose /Users/louis/Dropbox/Dev/Arduino/ArduinoJson_intToUintConversionTest/ArduinoJson_intToUintConversionTest.ino
/Applications/Teensyduino.app/Contents/Java/arduino-builder -compile -logger=machine -hardware /Applications/Teensyduino.app/Contents/Java/hardware -hardware /Users/louis/Library/Arduino15/packages -tools /Applications/Teensyduino.app/Contents/Java/tools-builder -tools /Applications/Teensyduino.app/Contents/Java/hardware/tools/avr -tools /Users/louis/Library/Arduino15/packages -built-in-libraries /Applications/Teensyduino.app/Contents/Java/libraries -libraries /Users/louis/Dropbox/Dev/Arduino/libraries -fqbn=teensy:avr:teensy36:usb=serial,speed=180,opt=o2std,keys=en-us -ide-version=10813 -build-path /var/folders/n6/l3jqlyms0r37qb1s_swjcqf00000gn/T/arduino_build_327691 -warnings=all -build-cache /var/folders/n6/l3jqlyms0r37qb1s_swjcqf00000gn/T/arduino_cache_65962 -verbose /Users/louis/Dropbox/Dev/Arduino/ArduinoJson_intToUintConversionTest/ArduinoJson_intToUintConversionTest.ino
Using board 'teensy36' from platform in folder: /Applications/Teensyduino.app/Contents/Java/hardware/teensy/avr
Using core 'teensy3' from platform in folder: /Applications/Teensyduino.app/Contents/Java/hardware/teensy/avr
Detecting libraries used...
/Applications/Teensyduino.app/Contents/Java/hardware/teensy/../tools/arm/bin/arm-none-eabi-g++ -E -CC -x c++ -w -g -Wall -ffunction-sections -fdata-sections -nostdlib -fno-exceptions -fpermissive -felide-constructors -std=gnu++14 -Wno-error=narrowing -fno-rtti -mthumb -mcpu=cortex-m4 -mfloat-abi=hard -mfpu=fpv4-sp-d16 -fsingle-precision-constant -D__MK66FX1M0__ -DTEENSYDUINO=153 -DARDUINO=10813 -DARDUINO_TEENSY36 -DF_CPU=180000000 -DUSB_SERIAL -DLAYOUT_US_ENGLISH -I/Applications/Teensyduino.app/Contents/Java/hardware/teensy/avr/cores/teensy3 /var/folders/n6/l3jqlyms0r37qb1s_swjcqf00000gn/T/arduino_build_327691/sketch/ArduinoJson_intToUintConversionTest.ino.cpp -o /dev/null -DARDUINO_LIB_DISCOVERY_PHASE
Alternatives for ArduinoJson.h: [ArduinoJson@6.18.0]
ResolveLibrary(ArduinoJson.h)
  -> candidates: [ArduinoJson@6.18.0]
/Applications/Teensyduino.app/Contents/Java/hardware/teensy/../tools/arm/bin/arm-none-eabi-g++ -E -CC -x c++ -w -g -Wall -ffunction-sections -fdata-sections -nostdlib -fno-exceptions -fpermissive -felide-constructors -std=gnu++14 -Wno-error=narrowing -fno-rtti -mthumb -mcpu=cortex-m4 -mfloat-abi=hard -mfpu=fpv4-sp-d16 -fsingle-precision-constant -D__MK66FX1M0__ -DTEENSYDUINO=153 -DARDUINO=10813 -DARDUINO_TEENSY36 -DF_CPU=180000000 -DUSB_SERIAL -DLAYOUT_US_ENGLISH -I/Applications/Teensyduino.app/Contents/Java/hardware/teensy/avr/cores/teensy3 -I/Users/louis/Dropbox/Dev/Arduino/libraries/ArduinoJson/src /var/folders/n6/l3jqlyms0r37qb1s_swjcqf00000gn/T/arduino_build_327691/sketch/ArduinoJson_intToUintConversionTest.ino.cpp -o /dev/null -DARDUINO_LIB_DISCOVERY_PHASE
Generating function prototypes...
/Applications/Teensyduino.app/Contents/Java/hardware/teensy/../tools/arm/bin/arm-none-eabi-g++ -E -CC -x c++ -w -g -Wall -ffunction-sections -fdata-sections -nostdlib -fno-exceptions -fpermissive -felide-constructors -std=gnu++14 -Wno-error=narrowing -fno-rtti -mthumb -mcpu=cortex-m4 -mfloat-abi=hard -mfpu=fpv4-sp-d16 -fsingle-precision-constant -D__MK66FX1M0__ -DTEENSYDUINO=153 -DARDUINO=10813 -DARDUINO_TEENSY36 -DF_CPU=180000000 -DUSB_SERIAL -DLAYOUT_US_ENGLISH -I/Applications/Teensyduino.app/Contents/Java/hardware/teensy/avr/cores/teensy3 -I/Users/louis/Dropbox/Dev/Arduino/libraries/ArduinoJson/src /var/folders/n6/l3jqlyms0r37qb1s_swjcqf00000gn/T/arduino_build_327691/sketch/ArduinoJson_intToUintConversionTest.ino.cpp -o /var/folders/n6/l3jqlyms0r37qb1s_swjcqf00000gn/T/arduino_build_327691/preproc/ctags_target_for_gcc_minus_e.cpp -DARDUINO_LIB_DISCOVERY_PHASE
/Applications/Teensyduino.app/Contents/Java/tools-builder/ctags/5.8-arduino11/ctags -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives /var/folders/n6/l3jqlyms0r37qb1s_swjcqf00000gn/T/arduino_build_327691/preproc/ctags_target_for_gcc_minus_e.cpp
Compiling sketch...
/Applications/Teensyduino.app/Contents/Java/hardware/teensy/../tools/precompile_helper /Applications/Teensyduino.app/Contents/Java/hardware/teensy/avr/cores/teensy3 /var/folders/n6/l3jqlyms0r37qb1s_swjcqf00000gn/T/arduino_build_327691 /Applications/Teensyduino.app/Contents/Java/hardware/teensy/../tools/arm/bin/arm-none-eabi-g++ -x c++-header -O2 -g -Wall -ffunction-sections -fdata-sections -nostdlib -MMD -fno-exceptions -fpermissive -felide-constructors -std=gnu++14 -Wno-error=narrowing -fno-rtti -mthumb -mcpu=cortex-m4 -mfloat-abi=hard -mfpu=fpv4-sp-d16 -fsingle-precision-constant -D__MK66FX1M0__ -DTEENSYDUINO=153 -DARDUINO=10813 -DARDUINO_TEENSY36 -DF_CPU=180000000 -DUSB_SERIAL -DLAYOUT_US_ENGLISH -I/Applications/Teensyduino.app/Contents/Java/hardware/teensy/avr/cores/teensy3 /var/folders/n6/l3jqlyms0r37qb1s_swjcqf00000gn/T/arduino_build_327691/pch/Arduino.h -o /var/folders/n6/l3jqlyms0r37qb1s_swjcqf00000gn/T/arduino_build_327691/pch/Arduino.h.gch
Using previously compiled file: /var/folders/n6/l3jqlyms0r37qb1s_swjcqf00000gn/T/arduino_build_327691/pch/Arduino.h.gch
/Applications/Teensyduino.app/Contents/Java/hardware/teensy/../tools/arm/bin/arm-none-eabi-g++ -c -O2 -g -Wall -ffunction-sections -fdata-sections -nostdlib -MMD -fno-exceptions -fpermissive -felide-constructors -std=gnu++14 -Wno-error=narrowing -fno-rtti -mthumb -mcpu=cortex-m4 -mfloat-abi=hard -mfpu=fpv4-sp-d16 -fsingle-precision-constant -D__MK66FX1M0__ -DTEENSYDUINO=153 -DARDUINO=10813 -DARDUINO_TEENSY36 -DF_CPU=180000000 -DUSB_SERIAL -DLAYOUT_US_ENGLISH -I/var/folders/n6/l3jqlyms0r37qb1s_swjcqf00000gn/T/arduino_build_327691/pch -I/Applications/Teensyduino.app/Contents/Java/hardware/teensy/avr/cores/teensy3 -I/Users/louis/Dropbox/Dev/Arduino/libraries/ArduinoJson/src /var/folders/n6/l3jqlyms0r37qb1s_swjcqf00000gn/T/arduino_build_327691/sketch/ArduinoJson_intToUintConversionTest.ino.cpp -o /var/folders/n6/l3jqlyms0r37qb1s_swjcqf00000gn/T/arduino_build_327691/sketch/ArduinoJson_intToUintConversionTest.ino.cpp.o
Compiling libraries...
Compiling library "ArduinoJson"
Compiling core...
Using precompiled core: /var/folders/n6/l3jqlyms0r37qb1s_swjcqf00000gn/T/arduino_cache_65962/core/core_6a8e06dec8be36068a20aa0373a23a9d.a
Linking everything together...
[... cut ...]
Global variables use 4196 bytes (1%) of dynamic memory, leaving 257948 bytes for local variables. Maximum is 262144 bytes.
/Applications/Teensyduino.app/Contents/Java/hardware/teensy/../tools/teensy_post_compile -file=ArduinoJson_intToUintConversionTest.ino -path=/var/folders/n6/l3jqlyms0r37qb1s_swjcqf00000gn/T/arduino_build_327691 -tools=/Applications/Teensyduino.app/Contents/Java/hardware/teensy/../tools -board=TEENSY36 -reboot -port=usb:14232140 -portlabel=[no_device] Serial -portprotocol=Teensy 
```

embedded-creations, 2021-06-29:
I found this in the code so I think it is at least meant to work, but haven't narrowed down the error yet:

https://github.com/bblanchon/ArduinoJson/blob/6.x/src/ArduinoJson/Numbers/convertNumber.hpp#L73

bblanchon, 2021-06-30:
Hi @embedded-creations,

Thank you very much for reporting this bug.
It's now fixed in the branch `6.x`.

Best regards,
Benoit

closed 2021-06-30 by bblanchon

bblanchon, 2021-07-03:
The fix is included in [ArduinoJson 6.18.1](https://github.com/bblanchon/ArduinoJson/releases/tag/v6.18.1)

commit ef757939 (2021-06-30): Fixed `JsonVariant::as<unsigned>()` (fixes #1601)