bblanchon/ArduinoJson #1603: exception while calling isNull() on empty array object

masterazz, 2021-06-30:
M5Stack Core esp32

i have an empty array in json input, looks like

"ar" : []

if i try to call 
["ar"][0].isNull() it returns true as expected, but if i call
["ar"][1].isNull() esp crashes with exeption

if i replace input with "ar" : [45]

all ["ar"][0].isNull() returns false and all the 
["ar"][1].isNull(), ["ar"][2].isNull(), ["ar"][3].isNull() returns true

<details open=""><summary>ArduinoJson Troubleshooter's report</summary> <ol><li>The issue happens at run time</li><li>The issue concerns deserialization</li><li><code class="language-plaintext highlighter-rouge">deserializeJson()</code> returns <code class="language-plaintext highlighter-rouge">Ok</code></li><li>Program crashes after calling <code class="language-plaintext highlighter-rouge">deserializeJson()</code></li><li>Program calls neither <code class="language-plaintext highlighter-rouge">strcmp()</code> nor <code class="language-plaintext highlighter-rouge">strcpy()</code></li><li>Replacing pointer with a <code class="language-plaintext highlighter-rouge">String</code> doesn’t solve the issue</li></ol></details>

bblanchon, 2021-07-01:
Hi @masterazz,

Thank you very much for reporting this bug.
I think I already fixed it in 898f42982675fd0987415eb7be8549e0a10d3c34.
The patch is included in 6.18.0.
Can you verify that you're up-to-date?

Best regards,
Benoit

masterazz, 2021-07-01:
thank you for your reply. yes i'm not up to date, i have used a fixed
version for offline development.

i will update it, seems it was the same problem

masterazz, 2021-07-01:
thanks, you are right