bblanchon/ArduinoJson #1606: Not Receiving Json Data

Bingsitfx, 2021-07-05:
I'm using arduinoJson version 5.13.5
I want to send and receive data from mega and nodemcu esp8266

This is  json data from mega to nodemcu using Serial2
`{"lux1":3440.719,"lux2":905.8619,"lux3":4562.809,"lux4":1223.858,"ratatemp1":23.61243,"ratatemp2":23.26854,"ratahum1":60.07039,"ratahum2":63.54948,"heater1":0,"humidifier1":0,"heater2":0,"humidifier2":0,"waktu":"14:16"}`

But nodemcu gives invalid
This nodemcu code
```
#include <SoftwareSerial.h>
#include <ArduinoJson.h>
//Variabel Software Serial (Rx, Tx)
SoftwareSerial DataSerial(14, 12);
SoftwareSerial DataSerialK(13, 15);
  Serial.begin(115200);
  DataSerial.begin(115200);
  DataSerialK.begin(115200);

void receive() {
StaticJsonBuffer<1000> JsonBuffer1;
    JsonObject& data1 = jsonBuffer1.parseObject(DataSerial);
    if(data1 == JsonObject::invalid())
      {
        Serial.println("Tidak menerima data mega");
        count_data = count_data + 1; 
        jsonBuffer1.clear();
        return;
      }
      lux1 = data1["lux1"];
      lux2 = data1["lux2"];
      lux3 = data1["lux3"];
      lux4 = data1["lux4"];
      temp1 = data1["ratatemp1"];
      temp2 = data1["ratatemp2"];
      hum1 = data1["ratahum1"];
      hum2 = data1["ratahum2"];
      heater1 = data1["heater1"];
      humi1 = data1["humidifier1"];
      heater2 = data1["heater2"];
      humi2 = data1["humidifier2"];
      const char* Waktu = data1["waktu"].as<const char*>();
      std::string waktu = Waktu;
      count_data = 0; 
}
}
```
nodemcu serial print :
`14:15:28.237 -> Tidak menerima data mega`

My wiring : 
D5 ->Tx2
D6 ->Rx2
D7 ->Tx3
D8 ->Rx3

Please help me, what should i do?

Pius171, 2021-07-05:
I don't think you are using nodemcu software serial correctly look up the nodemecu esp8266 software serial example in arduino

Bingsitfx, 2021-07-05:
> I don't think you are using nodemcu software serial correctly look up the nodemecu esp8266 software serial example in arduino

Where is it? my code is based on example arduino json parser.

bblanchon, 2021-07-05:
Hi @Bingsitfx,

Can you confirm that your program can successfully receive the JSON document without ArduinoJson?
BTW, why are you using ArduinoJson 5?

Best regards,
Benoit

Bingsitfx, 2021-07-05:
@bblanchon 
I can receive serial data with this code 
```
String data;
while(DataSerial.available()>0)
    {
      data += char(DataSerial.read());
    }
    data.trim();

    //uji data
    if(data !="")
    {
      count_data = 0;
      //Parsing data
      int index = 0;
      for(int i=0; i<= data.length(); i++)
      {
        char delimiter = '#' ;
        if(data[i] != delimiter)
          arrData[index] += data[i] ;
        else
         index++; 
      }
        rataldr1 = (arrData[0]).toFloat();
        rataldr2 = (arrData[1]).toFloat();
        temp1 = (arrData[2]).toFloat();
        temp2 = (arrData[3]).toFloat();
        hum1 = (arrData[4].toFloat());
        hum2 = (arrData[5]).toFloat();
        heater1 = (arrData[6]).toInt();
        humi1 = (arrData[7]).toInt();
        heater2 = (arrData[8]).toInt();
        humi2 = (arrData[9]).toInt();
        waktu = (arrData[10]);
```

The problem with the code above is that I have to parse the characters one by one. In addition there must be a `while` code that blocks the work of other code. I think that's why I want to use ArduinoJson.

Bingsitfx, 2021-07-06:
@bblanchon 
I try this code : 
```
    JsonObject& data1 = jsonBuffer1.parseObject(DataSerial);
    data1.prettyPrintTo(Serial);
```
This is serial output of nodemcu
11:33:52.906 -> {}Tidak menerima data mega

my wiring is still the same as before. 
D5 ->Tx2
D6 ->Rx2
D7 ->Tx3
D8 ->Rx3
GND ->GND

Pius171, 2021-07-06:
> > I don't think you are using nodemcu software serial correctly look up the nodemecu esp8266 software serial example in arduino
> 
> Where is it? my code is based on arduino json parser example

Go to examples in the arduino ide under the node MCU esp8266 section look for software serial

Pius171, 2021-07-06:
Have you tried using just hardware serial? Or better still use I2C. I have tried software serial with arduino and node MCU but I wasn't able to get them to communicate but hardware serial worked a bit fine for me. I have tried I2C also it also worked fine, maybe you should give it a try. On second thought did you use bidirectional logic level converter between arduino and node MCU arduino runs on 5v while node MCU 3V .

bblanchon, 2021-07-06:
> The serial data is like `245.555#22.33#45#` etc

Is this what you want to parse with ArduinoJson?

Bingsitfx, 2021-07-06:
> > The serial data is like `245.555#22.33#45#` etc
> 
> Is this what you want to parse with ArduinoJson?

No, thats my code before using ArduinoJson.
I want to parse this 
`"lux1":3440.719,"lux2":905.8619,"lux3":4562.809,"lux4":1223.858,"ratatemp1":23.61243,"ratatemp2":23.26854,"ratahum1":60.07039,"ratahum2":63.54948,"heater1":0,"humidifier1":0,"heater2":0,"humidifier2":0,"waktu":"14:16"`

I tried the latest version, but it still doesn't work.  "Invalid input" error appears. I tried between Uno (Serial) - Mega (Serial1) its normal, no error appears. What should i do?

Bingsitfx, 2021-07-06:
> Have you tried using just hardware serial? Or better still use I2C. I have tried software serial with arduino and node MCU but I wasn't able to get them to communicate but hardware serial worked a [...]

What is your suggestion, hardware serial or I2C?
What do you think if i use serial1 + i2c for data transfer? Because i want to send data to nodemcu then nodemcu retrieve firebase data and send to mega. Its need 2 serial right? Or just one? Im confuse

bblanchon, 2021-07-07:
@Bingsitfx, can you try the [ArduinoJson Troubleshooter](https://arduinojson.org/v6/troubleshooter/)?

Bingsitfx, 2021-07-08:
> @Bingsitfx, can you try the [ArduinoJson Troubleshooter](https://arduinojson.org/v6/troubleshooter/)?

I think the problem is arduino and nodemcu have difference on logic voltage, so its need logic level shifter right?

I will try this troubleshooter after the logic level shifter comes.

closed 2021-07-19 by bblanchon