bblanchon/ArduinoJson #1607: Arduino IDE v1.8.3 compiler errors due way of #include references

PaulskPt, 2021-07-05:
This is not a question. This is a report of what I have been confronted with. 
First: thank you for creating and publishing your ArduinJson repo.

I am not an experienced C/C++ programmer,
but I recently built some experience with C++ projects for the Raspberry Pi Pico.

I tried to compile and flash an example that uses the ThingESP library, which uses ArduinoJson.
First using Arduino IDE version 2.0.0 beta.  Total disaster. The IDE began to report the usual 'String ...' definitions as 'undefined'.
Then using the Arduino IDE version 1.8.13 (in MS Windows 10 latest): 
Many errors of 'No such file or directory' . Example:
../arduinojson/variant/VariantData.hpp:7.10:  '#include <ArduinoJson/Misc/SerializedValue.hpp>.
When I changed this into '#include ../Misc/SerializedValue.hpp" the compiler accepted.

In ArduinoJson there are various #include's using a path while the file referenced was in the same subfolder.
Example:  in the file: '/ArduinoJson/Variant/VariantSlot.hpp' is the include: '#include <ArduinoJson/Variant/VariantContent.hpp>'.
I changed it into: '"include VariantContent.hpp".

Recently I had the same experiences when compiling C++ projects for the Raspberry Pi Pico (pico-sdk). Using relative 
references like "../<filename.h>" or "../../../../<filename.h>" worked instead of  "<thisFolder>/<filename.h>".

Regards,
Paulsk

bblanchon, 2021-07-05:
Hi Paulsk,

How did you install the library?

Best regards,
Benoit

bblanchon, 2021-07-05:
PS: ArduinoJson uses an absolute path to workaround #1072

PaulskPt, 2021-07-05:
Hi Benoit, mci.
I downloaded the zip file of your repo. Unpacked it. Initially I copied it to the 'user' libraries folder in:
"c:\Users\<username>\Documents\Arduino\libraries"
but later I decided to copy it to:
C:\Users\<username>\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.1\libraries\

Excuse me but I think it doesn't matter so much where it is installed. The errors are relative related within the ThingESP library.
The errors are not related to files outside of ThingESP.

Regarding your reference to the workaround.: in my case the errors were not related to a long filename path.

[image]

PaulskPt, 2021-07-05:
Hi Benoit, mci.
I downloaded the zip file of your repo. Unpacked it. Initially I copied it to the 'user' libraries folder in:
"c:\Users\<username>\Documents\Arduino\libraries"
but later I decided to copy it to:
C:\Users\<username>\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.1\libraries\

Excuse me but I think it doesn't matter so much where it is installed. The errors are relative related within the ThingESP library.
The errors are not related to files outside of ThingESP.

Regarding your reference to the workaround.: in my case the errorrs were not related to a long filename path.

[image]

PaulskPt, 2021-07-05:
I Checked: I have ThingESP also in the Tools > Manage Libraries.

[image]

bblanchon, 2021-07-06:
@PaulskPt, how did you solve your issue?

PaulskPt, 2021-07-06:
Hi Benoît, by error I clicked the 'close' button. I edited every file until there were no more 'file not found' errors. I sometimes also edited other files that were in a same subdirectory. I edited for several hours. Maybe I edited 50 files (or more). If you like I can make a .zip and upload it to you (if that is allowed by GitHub, I don't know). Now I have some other type of errors in the sketch to deal with.

Foinard, 2021-07-07:
Same issue

bblanchon, 2021-07-07:
@PaulskPt, if you don't want to install ArduinoJson via the library manager, you can download the single file distribution.
See [installation instructions](https://arduinojson.org/v6/doc/installation/) for details.

@Foinard, please elaborate:

* Are you using Arduino IDE too?
* Did you download the zip the same way?
* Did it work with older versions of the library?
* What did you try to fix this issue?

PaulskPt, 2021-07-07:
@bblanchon  Thank you. OK, I already wrote above that I installed the library via the standard way.

bblanchon, 2021-07-07:
> I downloaded the zip file of your repo. Unpacked it. 

@PaulskPt, is this what you call the standard way?

PaulskPt, 2021-07-07:
[image]

bblanchon, 2021-07-08:
I understand that you installed the ThingESP via the Library Manager, but what about ArduinoJson?

PaulskPt, 2021-07-08:
[image]

Foinard, 2021-07-08:
Solved !
I tried to install it with the library manager and it worked !

bblanchon, 2021-07-09:
@PaulskPt, I bet there is something wrong with your installation.
Can you try to uninstall and reinstall?

closed 2021-07-27 by bblanchon