bblanchon/ArduinoJson #1608: Bug in garbageCollect() on ESP32

Alonzo67, 2021-07-07:
Hello
I am trying to test a small program on a ESP32 to record four values in a Json buffer.
When I do a garbageCollect() the first 3 values ​​of the buffer are modified and the size of the array is reset.
Why ?
Tanks for your help

#include <ArduinoJson.h>  // ArduinoJson by Benoit Blanchon Ver 6.18.0

StaticJsonDocument<4300> jsonBuffer;                 // Buffer static contenant le JSON courant 
JsonArray timestamp = jsonBuffer.createNestedArray("timestamp");
JsonArray hist_t = jsonBuffer.createNestedArray("t");
JsonArray hist_h = jsonBuffer.createNestedArray("h");
JsonArray hist_c = jsonBuffer.createNestedArray("c");

char json[4300];                                   // Buffer pour export du JSON - JSON export buffer

void setup() {
  // put your setup code here, to run once:
Serial.begin(115200);
}

void loop() {
  // put your main code here, to run repeatedly:
  timestamp.add("1625660330");
  hist_t.add("33.4");
  hist_h.add("56.5");
  hist_c.add("38");
  Serial.print("Mémoire : ");
  Serial.println(jsonBuffer.memoryUsage());
  Serial.println(hist_t.size());
  serializeJson(jsonBuffer, json); 
  Serial.println(json);

   if ( hist_t.size() > 60 ) {
      timestamp.remove(0);
      hist_t.remove(0);
      hist_h.remove(0);
      hist_c.remove(0);
     Serial.println("Remove");
     }

   if(jsonBuffer.memoryUsage() > 4200){
      jsonBuffer.garbageCollect();
      Serial.println("Garbage Collect ");
     }
  
 delay(500);
}

bblanchon, 2021-07-08:
Hi @Alonzo67,

Internally, [JsonDocument::garbageCollect()](https://arduinojson.org/v6/api/jsondocument/garbagecollect/) makes a temporary copy of the document. Since you use a [StaticJsonDocument](https://arduinojson.org/v6/api/staticjsondocument/), this copy is placed on the stack. The problem is that the document is huge (4300 bytes) and probably overflows the stack, hence the undefined behavior. Switching to a [DynamicJsonDocument](https://arduinojson.org/v6/api/dynamicjsondocument/) fixes this issue.

Then, there is a second problem in this program. As the documentation says "Every reference (JsonArray, JsonObject, or JsonVariant) acquired before calling garbageCollect() is invalidated", so `timestamp`, `hist_t`, `hist_h`, and `hist_c` are all dangling after the call to `garbageCollect()`. One solution is to reassign each reference after the call.

Here is the fixed program:

```c++
#include <ArduinoJson.h>  // ArduinoJson by Benoit Blanchon Ver 6.18.0

DynamicJsonDocument jsonBuffer(4300);                 // Buffer static contenant le JSON courant 
JsonArray timestamp = jsonBuffer.createNestedArray("timestamp");
JsonArray hist_t = jsonBuffer.createNestedArray("t");
JsonArray hist_h = jsonBuffer.createNestedArray("h");
JsonArray hist_c = jsonBuffer.createNestedArray("c");

char json[4300];                                   // Buffer pour export du JSON - JSON export buffer

void setup() {
  // put your setup code here, to run once:
Serial.begin(115200);
}

void loop() {
  // put your main code here, to run repeatedly:
  timestamp.add("1625660330");
  hist_t.add("33.4");
  hist_h.add("56.5");
  hist_c.add("38");
  Serial.print("Mémoire : ");
  Serial.println(jsonBuffer.memoryUsage());
  Serial.println(hist_t.size());
  serializeJson(jsonBuffer, json); 
  Serial.println(json);

   if ( hist_t.size() > 60 ) {
      timestamp.remove(0);
      hist_t.remove(0);
      hist_h.remove(0);
      hist_c.remove(0);
     Serial.println("Remove");
     }

   if(jsonBuffer.memoryUsage() > 4200){
      jsonBuffer.garbageCollect();
      Serial.println("Garbage Collect ");
      timestamp = jsonBuffer["timestamp"];
      hist_t = jsonBuffer["t"];
      hist_h = jsonBuffer["h"];
      hist_c = jsonBuffer["c"];
     }
  
 delay(500);
}
```

Best regards,
Benoit

Alonzo67, 2021-07-08:
It works well. thanks a lot for your help.

Merci beaucoup .

closed 2021-07-08 by Alonzo67

bblanchon, 2021-07-08:
You're welcome, @Alonzo67.
Thank you for using ArduinoJson.
Good luck with your project!