bblanchon/ArduinoJson #1609: Question:  Unknown List Length - Best Approach?

lbussy, 2021-07-07:
Hi Benoit:

I am trying to return a serialized JSON string containing an array of JSON lists (or is it a list of arrays?) and I was wondering if you could recommend the most effective approach?  The specific use case here is returning a JSON document containing scanned Access Points, and I won't know how many will be scanned for sure till after the scan is done.  The structure of each would be something like this:

``` json
{
    "apname": "xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx",
    "rssi": -100,
    "secure": false
}
```

I guess I could always hard-code it to a max of 32 detected access points, but that seems wasteful.

bblanchon, 2021-07-08:
Hi Lee,

You must hardcode the maximum size because ArduinoJson 6 doesn't offer any alternative, and I don't think that's a bad thing.

I know it seems like a waste, but is it really?
Imagine that you didn't have to allocate this space; what would you do with the saved RAM?
You can't use it; otherwise, there may not be enough remaining RAM to store all access points.
Your program must be able to report up to 32 access points, right? So, one way or another, these 32 slots must be available.

If you use ArduinoJson as intended (i.e., only for serialization/deserialization), this cannot be a problem because the life of the `JsonDocument` is so short that it releases the memory instantly.
If you don't use ArduinoJson as intended (e.g., you use `JsonDocument` to store the state of your application), you can call `shrinkToFit()` to release the unused memory.

I plan on adding the elastic capacity back in v7 (#1074), but the overhead will be significant.
Remember what happened with v6.6: I implemented the elastic `JsonDocument` and then reverted it in v6.7 because the overhead in memory consumption and code size was unacceptable.

I know the idea of the elastic `JsonDocument` is seducing, but it's a lie.
The fixed capacity is what makes ArduinoJson so lean and fast.

Best regards,
Benoit

lbussy, 2021-07-08:
Thanks for the reply.  Actually the JSON was bound for a web page not to be retained, but I can live with static allocation.  If nothing else I can just grab the top X strengths and report those.

I just wanted to make sure before I went and did something that I was not reinventing the wheel.

closed 2021-07-08 by lbussy