bblanchon/ArduinoJson #1610: Any Release on Ubuntu? libarduinojson-dev

HappySamuel, 2021-07-08:
Hi
Will there be any deb release on Linux?

Best,
Samuel

bblanchon, 2021-07-09:
Hi Samuel,

Thank you very much for this question.

No, there isn't any plan for a .deb package.
However, I would accept a PR that adds this package to the CI.
We can also leave this issue open to see if more people are interested.

To be honest, I doubt it would be downloaded by many users.
If you want to use ArduinoJson on Linux, I recommend that you download the [single header](https://github.com/bblanchon/ArduinoJson/releases/download/v6.18.1/ArduinoJson-v6.18.1.h) and save it with your project files.

Best regards,
Benoit

HappySamuel, 2021-07-12:
Hi @bblanchon 

Thank you for the single header. I am going to try it out.

Best
Samuel

bblanchon, 2021-09-09:
Is anyone interested in this Unbuntu package?
If not, I'll close this issue.

closed 2021-09-09 by HappySamuel