bblanchon/ArduinoJson #1611: Need help by Changing over from v5 to v6

miloit, 2021-07-20:
Hello,
how do I need to rewrite this part of code?

```
DynamicJsonBuffer jsonBuffer(JSON_BUFFER_DEVICE_MAPPING_CONFIG_FILE);
  JsonObject& json = jsonBuffer.createObject();

  for(int index = 0; index < m_cfgDeviceMapping.size(); index++)
  {
    const DeviceMapping* mapping = m_cfgDeviceMapping.get(index);
        
    if(strlen(mapping->name) != 0)
    {
      Serial.print(F("Preparing to write device mapping config file index "));
      Serial.println(String(index));
        
      JsonObject& deviceMappingEntry = json.createNestedObject(String(index));
  
      deviceMappingEntry[JSON_KEY_DEVICEMAPPING_NAME] = mapping->name;
      deviceMappingEntry[JSON_KEY_DEVICEMAPPING_TYPE] = (int)mapping->type;
      deviceMappingEntry[JSON_KEY_DEVICEMAPPING_LED]  = (int)mapping->ledNumber;
    }
    else
    {
      Serial.print(F("Removing device mapping config file index "));
      Serial.println(String(index));
    }
  }
  
  if(!m_deviceMappingConfigFile.write(&json))
  {
    onFileWriteError();
  } 
  else
  {
    m_cfgDeviceMappingDirty = false;
  }
```

bblanchon, 2021-07-21:
Hi @miloit,

Your translation seems correct.
Does it work?

Best regards,
Benoit

miloit, 2021-07-21:
> Hi @miloit,
> 
> Your translation seems correct.
> Does it work?
> 
> Best regards,
> Benoit

Can't test it because i try to translate this code but I am complety lost...... Can you help me ? @bblanchon 

```
bool success = false;

  char fileBuffer[MAX_SIZE_MAIN_CONFIG_FILE];

  if(m_mainConfigFile.read(fileBuffer, MAX_SIZE_MAIN_CONFIG_FILE))
  {
    DynamicJsonBuffer jsonBuffer(JSON_BUFFER_MAIN_CONFIG_FILE);
    JsonObject& json = jsonBuffer.parseObject(fileBuffer);

    if (json.success()) 
    {
      Serial.println(F("Main config data successfully parsed."));
      Serial.print(F("JSON length is ")); Serial.println(json.measureLength());     
      printMainConfigFile(json);
      Serial.println(F(""));

      if(json.containsKey(JSON_KEY_HOST) && json.containsKey(JSON_KEY_WIFI_SSID) && json.containsKey(JSON_KEY_WIFI_PSK) && 
         json.containsKey(JSON_KEY_MQTT_SERVER) && json.containsKey(JSON_KEY_MQTT_SERVERUSER) && json.containsKey(JSON_KEY_MQTT_SERVERPASSWORD) && json.containsKey(JSON_KEY_MQTT_STATUS_TOPIC) && json.containsKey(JSON_KEY_MQTT_TEST_TOPIC) && json.containsKey(JSON_KEY_MQTT_WILL_TOPIC) &&
         json.containsKey(JSON_KEY_LED_COUNT) && json.containsKey(JSON_KEY_LED_PIN))
      {
        Serial.println(F("Config data is complete."));

        setHost(json[JSON_KEY_HOST]);
        setWifiSSID(json[JSON_KEY_WIFI_SSID]);
        setWifiPSK(json[JSON_KEY_WIFI_PSK]);
        setMqttServer(json[JSON_KEY_MQTT_SERVER]);
        setMqttServerUser(json[JSON_KEY_MQTT_SERVERUSER]);
        setMqttServerPassword(json[JSON_KEY_MQTT_SERVERPASSWORD]);
        setMqttStatusTopic(json[JSON_KEY_MQTT_STATUS_TOPIC]);
        setMqttTestTopic(json[JSON_KEY_MQTT_TEST_TOPIC]);
        setMqttWillTopic(json[JSON_KEY_MQTT_WILL_TOPIC]);
        setNumberOfLeds(json[JSON_KEY_LED_COUNT]);
        setLedDataPin(json[JSON_KEY_LED_PIN]);
        setLedBrightness(json[JSON_KEY_LED_BRIGHTNESS]);

        success = true;
      }
    } 
    else 
    {
      Serial.println(F("Could not parse config data."));
    }
  }
  else
  {
    Serial.println(F("Creating default main config file."));
    resetMainConfigData();
    writeMainConfigFile();
  }

  return success;
```

bblanchon, 2021-08-15:
@miloit, did you solve your problem?

closed 2021-08-21 by bblanchon