bblanchon/ArduinoJson #1612: How to add char buffer with length to JsonArray?

bhcuong2008, 2021-07-20:
Hi,

This is not a bug report. I just want to know how to add char buffer with length to JsonArray?
Currently JsonArray supports add(char*), it's for c-string with null terminated. I have a buffer of char, without null terminated. So how could I do that? 

I think Json should support add string with length. Json will automatically append null char to the end after copying from the source char buffer.

Thank you so much.
Cuong.

bblanchon, 2021-07-21:
Hi Cuong,

You can use `std::string_view` like so:

```c++
doc["key"] = std::string_view("valueXXXX", 5);
```

ArduinoJson supports `std::string_view` since version 6.18.1.

Best regards,
Benoit

bhcuong2008, 2021-07-21:
Thank @bblanchon ,

if using `doc["key"] = std::string_view("valueXXXX", 5);` then does ArduinoJson use copy mode or reference mode?

Besides that, std::string_view seems to be not supported in ESP32 Arduino. std::string_view supported from C++17. ESP32 Arduino just supports C++11.

Could you guide me a simple way to add support in traditional way, like this:
```
char charBuf[10];                     // not null terminated
doc.add(charBuf, 10)     // as copy mode, doc as JsonArray
```

Thank you.

bblanchon, 2021-07-21:
> Could you guide me a simple way to add support in traditional way...

This would be a significant work and I would not be able to guarantee the quality of the implementation, so I don't want to follow this route.

Please read the article [How to use string_view on ESP32?](https://arduinojson.org/v6/how-to/use-string-view-on-esp32/)
I just wrote it for you, @bhcuong2008; I hope it helps.

bhcuong2008, 2021-07-21:
I've read it. It's really helpful. Thank you very much, @bblanchon 

string_view is generic, better than char array with length. 

In the case of string_view, `doc.add(std::string_view(charBuf, 10))` is in copy mode? It means that if I destroy charBuf, it still exists in doc?

Thank @bblanchon again.

bblanchon, 2021-07-21:
Yes, ArduinoJson stores `string_view` content by copy; I'm sorry I forgot to answer this question.

bhcuong2008, 2021-07-21:
Thank @bblanchon :)

bhcuong2008, 2021-07-21:
I have one more question. After saving to the doc, now I want to access it, is it like as const char* like this,

`buf = doc[0].as<const char*>();`

bblanchon, 2021-07-22:
Yes, and the returned string will be zero-terminated.

bhcuong2008, 2021-07-25:
Thank you, @bblanchon

bblanchon, 2021-07-26:
You're welcome, @bhcuong2008!
Thank you for using ArduinoJson.
Good luck with your project.

closed 2021-07-26 by bblanchon