bblanchon/ArduinoJson #1613: Deserialization use JsonPair to get key and value

vanderareosa, 2021-07-22:
I'm trying to deserialize the value that arrives by BLE where I don't know who is the key and also its value. I can get the key but the value print is empty. The idea is to receive all incoming messages either via HTTP or BLE, handle the message and send it as char* to another method, so I need to remove the key and send only the content to another method. Below my code

void handlerBLEMessage(const char* msg){
    const char* value;
    const char* key; 
    DynamicJsonDocument doc(600);                   
    DeserializationError error =  deserializeJson(doc, msg);
    if (error){
      Serial.print(F("deserializeJson() at handlerBLEMessage failed: "));
      Serial.println(error.f_str());
    }else{    
      JsonObject root = doc.as<JsonObject>();
      for (JsonPair kv : root) {
          key = kv.key().c_str();
          value = kv.value().as<char*>();
          Serial.printf("key[%s] \n",key);
          Serial.printf("value[%s] \n",value); //-- the value print empty
      }

      if (strcmp(key, "status"))
              saveConfigParams(value, key);              
     else
              powerDevice();

  }
}        

the value of msg is:  
{
  "temperature": {
    "powerByTemperature": true,
    "temperatureOn": 36,
    "temperatureOff": 28
  }
}

key value print [temperature]  ... its okay
value print nothing.

I will appreciate it for some help.

bblanchon, 2021-07-22:
Hi @vanderareosa,

`kv.value().as<char*>()` only works if the value is a string; in all other cases, it returns null.
In your case, none of the values are strings: the first is a boolean, and the two others are numbers.

To print the value of a [`JsonVariant`](https://arduinojson.org/v6/api/jsonvariant/) whatever its type, you must call [`serializeJson()`](https://arduinojson.org/v6/api/json/serializejson/), like so:

```c++
serializeJson(kv.value(), Serial);
```

By the way, `as<char*>()` is deprecated since 6.18; instead you must use `as<const char*>()`.
Also, you don't need to call `strcmp()`: you can use the `==` with `.key()`.

Best regards,
Benoit

vanderareosa, 2021-07-22:
perfect, but how can pass the value as const char * to my saveConfigParams(value, key) that wait a const char* as parameter? this method handle message that come from Stream Firebase too, that is const char*.

vanderareosa, 2021-07-22:
Thanks so much! Works perfectly.

closed 2021-07-22 by vanderareosa