bblanchon/ArduinoJson #1614: Deserialize a JSON payload that contains single quotes in JSON string

toblum, 2021-07-23:
Hello all,

I looked through countless issues but didn't find a answer to my question, so I guess I have to ask this one :-)

I'm trying to parse a JSON payload using deserializeJson() directly from a stream on ESP32. The JSON payload originates from the Microsoft Graph API and looks like this:

```JSON
{
    "@odata.context": "https://graph.microsoft.com/v1.0/$metadata#users('fhg7226a-0012-4d8c-9f6a-46c58g757a02')/events(subject,start,end,location,bodyPreview)",
    "value": [{
            "@odata.etag": "W/\"DRhq7FgfZP+4/wABwT/0eQ==\"",
            "id": "AAMkAGI2OTRhMz=",
            "subject": "Som etext"
        }
	],
    "@odata.nextLink": "https://graph.microsoft.com/v1.0/me/events?%24select=subject%2cstart%2cend%2clocation%2cbodyPreview&%24top=5&%24skip=5"
}
```
This fails with "InvalidInput error during deserialization".

I didn't go through all the JSON specification in detail, but as far as I understand it, this is perfectly valid JSON. At least jsonlint.com and others says it is. :-) I noticed that ArduinoJSON treats single and double quotes for object keys equally and I guess this is what's causing this issue.

Is there a way to change the behaviour of ArduinoJSON so that this payload can be deserialized? I could escape the single quotes in the payload before deserializing, but that would be only the second best option as I want to read directly from the stream as the payloads can get big.

I look forward to hearing from you and am curious about your answer.

Many greetings
Tobias

bblanchon, 2021-07-25:
Hi Tobias,

This JSON document is valid and ArduinoJson parses it successfully as you can see in this online demo:
https://wandbox.org/permlink/5pUPu3TeLzHL1OQB

The [ArduinoJson Troubleshooter](https://arduinojson.org/v6/troubleshooter) should help you find the problem.
If not, please share a small code sample (max 50 lines) reproducing the problem.

Best regards,
Benoit

toblum, 2021-07-25:
Hi @bblanchon,

thank you for your quick reply. And: you are 100% right. It really does work absolutely as it should.

For my own defence: I didn't make it easy for myself and checked the behaviour very carefully. And I wasn't the only one who noticed that the deserialisation only worked when the single quotes were replaced beforehand. As a reference, the original ticket was this: https://github.com/toblum/ESPTeamsPresence/issues/19

Today, however, I can no longer reconstruct the behaviour. Everything works as it should. I therefore assume that the problem originates elsewhere and that the observation with the single quotes was only coincidental. I'll have to look into this more closely if the problem occurs again.

Many greetings
Tobias

closed 2021-07-25 by toblum

bblanchon, 2021-07-26:
Hi Tobias,

In the screen capture posted by @pduck27, I see `d2` before the JSON document.
This looks like a chunk start as defined by [chunked transfer encoding](https://en.wikipedia.org/wiki/Chunked_transfer_encoding).
With HTTP/1.1, the server can decide to enable or disable it on a request basis (for example, when the response is in cache), which explains why the problem is not reproducible.

To fix this issue, you must either:

* Handle chunked transfer encoding, before passing the stream to `deserializeJson()` (good luck with that)
* Downgrade to HTTP/1.0 (piece of cake)

The [ArduinoJson Troubleshooter](https://arduinojson.org/v6/troubleshooter/) should have told you about that.

Best regards,
Benoit

toblum, 2021-07-26:
Hi @bblanchon,

indeed! I missed these signs completely. Maybe because I [switched to HTTP/1.0](https://github.com/toblum/ESPTeamsPresence/blob/41d1980819130d8e429647f2bddb53c438c5e422/src/request_handler.h#L39) some time ago and had therefore dismissed this possibility in my mind.

And now it all makes sense: @pduck27 must use a older fork of the project (can see this from the serial messages that are not in my code) without the HTTP/1.0 fix. When I tested it, I boiled everything down to the absolute minimum and also forgot to set HTTP/1.0. The fix @pduck27 proposed required to use .getString() instead of .getStream() and that solved the problem, not the replacement of the single quotes.

Shame on me! I should've known better. Never trust a finding that you didn't have by yourself! Always start from scratch. :-)

Thank you again!

Greetings!

---

BTW: I tried the troubleshooter and went that route:
<details open=""><summary>ArduinoJson Troubleshooter's report</summary> <ol><li>The issue happens at run time</li><li>The issue concerns deserialization</li><li><code>deserializeJson()</code> returns <code>InvalidInput</code></li><li>Input comes from a stream</li><li><code>jsonlint</code> says the document is valid</li><li>Adding a buffer doesn’t solve the issue</li><li>Input’s first byte doesn’t suggest a BOM</li></ol></details>

So I didn't stumble over the issue, because server answered without chunking in step 5 and JSON was valid. Not sure why it doesn't happen all the time. Could depend on the edge server you're hitting.

bblanchon, 2021-07-27:
Hi Tobias,

The Troubleshooter would have warned you about CTE if you told him that the input comes from an HTTP response.
Indeed, "stream" was an equally good answer in your case; I wonder how I could make it more obvious that "HTTP response" is the appropriate choice...

Anyway, I'm glad you finally found the explanation!

Best regards,
Benoit

toblum, 2021-07-27:
Thank you very much for the support!