bblanchon/ArduinoJson #1615: serializeJson(doc, mqttClient) for large doc size

steven-higgins, 2021-07-25:
My apologies for posting here, its not a bug report but I cannot find a solution to this problem in the documentation, so I am posting this question here:

I am using serializeJson(doc, mqttClient) to successfully send messages to mosquitto mqtt. Specially I use (from your examples):

mqttClient.beginMessage(topic);
serializeJson(doc, mqttClient);
mqttClient.endMessage();

This works up to a buffer size of 256 bytes and serializeJson(doc, Serial) illustrates doc is correctly formatted even for larger buffer sizes. The mosquitto broker is using default settings i.e. maximum payload size of 268435455 bytes, so I believe that the problem is that only 256 bytes are being sent and not that the broker is truncating the messages. The broker reports exactly 256 bytes. 

Arduino: 1.8.15 (Linux), Board: "Arduino NANO 33 IoT". arduinojson 6.18.0.  arduinomqttclient 0.1.5

many thanks, Steve

bblanchon, 2021-07-26:
Hi Steve,

I just checked the source of ArduinoMqttClient, and it does have a 256-byte tx buffer:
https://github.com/arduino-https://github.com/arduino-libraries/ArduinoMqttClient/blob/0.1.5/src/MqttClient.cpp#L35

However, **it seems** that you can bypass this buffer, by specifying the message size to `beginMessage()`:
https://github.com/arduino-libraries/ArduinoMqttClient/blob/0.1.5/src/MqttClient.cpp#L174

If that's true, the following would fix your issue:

```c++
mqttClient.beginMessage(topic, measureJson(doc));
serializeJson(doc, mqttClient);
mqttClient.endMessage();
```

If this doesn't work, please contact the authors of the library.
In any case, please keep me informed of your progress, so I can update the documentation on arduinojson.org.

Best regards,
Benoit

steven-higgins, 2021-07-26:
Hi Benoit,

thanks for looking into this. I will try your suggestion and report back - I am away from the hardware at the moment. 

In the meantime I have a work around which is to create the large doc and then partition it into smaller docs for sending. This means a few more lines of code are executed on the Arduino, but this simplifies later processing on node-red. So I might stay with this work around for this project.

all the best
Steve

bblanchon, 2021-08-15:
Hi Steve,

Did you find the time to check if passing the message size to `beginMessage()` allows sending large messages?

Best regards,
Benoit

closed 2021-09-06 by bblanchon