bblanchon/ArduinoJson #1616: Serialize milisecond epoch (appending 3 ceros at end)

EmilianoQS, 2021-07-28:
Hello. This is not an issue, but a question.

I have to serialize a milisecond epoch, but I don't need (nor have) milisecond accuracy. I have a second accuracy epoch and convert it to milisecond.

To do that conversion I have 2 options. Treat that unsigned long variable as a number and multiply it by 1000 (must use 64bit variables), or treat it like a char (or string) and just append 3 ceros at the end.

So the question is: 
**Is there any way to serialize a char like if it was a number (without quotes "" ) but without converting it to number?**

Thanks!

bblanchon, 2021-07-29:
Hi @EmilianoQS,

You can mark a string as [serialized()](https://arduinojson.org/v6/api/misc/serialized/) so that [serializeJson()](https://arduinojson.org/v6/api/json/serializejson/) doesn't add quotes nor escape special characters.
This is explained in the [Serialization Tutorial](https://arduinojson.org/v6/doc/serialization/) 😞

Best regards,
Benoit

EmilianoQS, 2021-07-29:
Thanks Benoit! 

I searched for it, but it seems not so good. Sorry.

Best regards!

closed 2021-07-30 by bblanchon