bblanchon/ArduinoJson #1617: Retrieving values within arrays

wvthoog, 2021-07-29:
I've created a PHP script that spits out some information about Google Analytics in JSON format and i want to interpret it with an ESP32 but having some issues doing so. 

I'm using HTTPClient to retrieve the payload String from my website (String payload = http.getString();) which results in the following output:

```
{
  "0": {
    "country": "NL",
    "city": "Rotterdam"
  },
  "1": {
    "country": "BE",
    "city": "Brussel"
  },
  "2": {
    "country": "FR",
    "city": "Parijs"
  },
  "3": {
    "country": "DE",
    "city": "Berlijn"
  },
  "activeusers": "4",
  "totalsessions": "1768"
}
```

when there is no active user connected it results in this 

```
{
"activeusers":0
}
```

So when i use the ArduinoJson Assistant it suggests i use StaticJsonDocument. This works but is static, i need it for it to be dynamic and probably need a for loop to go through each key and retrieve it's value. Tried this:

```
        for (int i = 0; i < doc.size(); i++)
        {
          String root_country = doc[i]["country"];
          String root_city = doc[i]["city"];

          Serial.println(root_country);
          Serial.println(root_city);
        }
```

The main problem is (i think) that there are arrays within the main array which increment (0, 1, 2 etc) from which i want to retrieve the country and city values. 

A bit stuck here. Hope someone can point me in the right direction.

Thanks

bblanchon, 2021-07-30:
Hi,

You cannot use an integer as an index for an object; you must use a string instead.
Therefore, you could fix your problem by replacing `doc[i]` with `doc[String(i)]`, but it would be quite inefficient.

I think your problem roots in the JSON file layout: users should be in an array, not in an object.
Since you have the hand on this code, I suggest that you change the layout to something more convenient, like so:

```json
{
  "activeusers": [
    {
      "country": "NL",
      "city": "Rotterdam"
    },
    {
      "country": "BE",
      "city": "Brussel"
    },
    {
      "country": "FR",
      "city": "Parijs"
    },
    {
      "country": "DE",
      "city": "Berlijn"
    }
  ],
  "totalsessions": 1768
}
```

This way you can enumerate active users in an efficient way, like so:

```c++
for (JsonObject user : doc["activeusers"].as<JsonArray>()) {
  Serial.println(user["country"].as<const char*>());
  Serial.println(user["city"].as<const char*>());
}
```

Notice that I change `totalsession` into a number and avoided the costly conversion to `String` in the loop.

Another benefit of this new JSON layout is that it allows using the "deserialization in chunks" technique if you don't have enough RAM to store the complete document.

See also:

* [Deserialization tutorial](https://arduinojson.org/v6/doc/deserialization/)
* [JsonArray::begin() / JsonArray::end()](https://arduinojson.org/v6/api/jsonarray/begin_end/)
* [How to deserialize a very large document?](https://arduinojson.org/v6/how-to/deserialize-a-very-large-document/)

Best regards,
Benoit

wvthoog, 2021-07-30:
Exactly the answer is was hoping for. Rewrote the PHP script to format it as you described and now all is well. 

```
        StaticJsonDocument<384> doc;
        
        DeserializationError error = deserializeJson(doc, payload);
        
        if (error) {
          Serial.print(F("deserializeJson() failed: "));
          Serial.println(error.f_str());
          return;
        }

        for (JsonObject user : doc["activeusers"].as<JsonArray>()) {
        Serial.println(user["country"].as<const char*>());
        Serial.println(user["city"].as<const char*>());
      }
        Serial.println(doc["totalsessions"].as<int>());
        Serial.println(doc["activeusers"].size());
```
Works like a charm. Thank you so much.

Issue can be closed

closed 2021-07-31 by bblanchon