bblanchon/ArduinoJson #1618: Create union of std::string and DynamicJsonDocument?

bhcuong2008, 2021-07-30:
Hi,

This is my question, not issue.
Currently I use std::string to store Json string. Then when I need to process data, I deserialize it to JsonDocument. It's take processing time, and RAM, and may cause heap fragment.

So I want to have an union like this. When processing done, I want to release JsonDocument to zero.
```
union {
   std::string s;
   DynamicJsonDocument doc;
}
```

But at time of definition, it needs to define buffer size, such as DynamicJsonDocument doc(4096);
For this reason, I will use pointer like this
```
union {
std::string s;
DynamicJsonDocument *doc;
}
```

When init data, I do this: 
`doc = new DynamicJsonDocument(4096);`

When done, I do:
```
delete doc;
doc = NULL
```

Is this way ok?

Thank you so much.

bblanchon, 2021-07-31:
Hi @bhcuong2008,

You can use a `DynamicJsonDocument` as a union member in C++11.
However, you must explicitly call the constructor with a placement `new`.
(see [C++11 FAQ on unions](https://www.stroustrup.com/C++11FAQ.html#unions))

To be honest, I never tried, and I don't think I event will.
A `JsonDocument` is already a discriminated union, so why would I embed it in another union?
Instead, I would store the string in the `JsonDocument` by calling `doc.set("example")`.

Best regards,
Benoit

bhcuong2008, 2021-08-01:
Thank @bblanchon ,

How about this, when do delete, is heap memory allocated by JsonDocument 4096, together with doc itself, is freed?
```
DynamicJsonDocument *doc =  new DynamicJsonDocument(4096);
delete doc; // after use
```

bhcuong2008, 2021-08-02:
Thank u for your info.

closed 2021-08-02 by bhcuong2008