bblanchon/ArduinoJson #1619: How to determine Json buffer if message size known?

bhcuong2008, 2021-07-30:
Hi,

I have already read ArduinoJson website about how to determine Json document buffer size. But it does not still answer my concern. So I ask here. I use ArduinoJson 6.18.3

If message length is known, its json structure, max level may be guessed. So how do I make Json buffer suitable. All is copy mode, stored in Json document. 

Currently, I use fixed buffer size for all message. So it's not optimal. Some just needs 256 bytes. Some needs several KBs.

Thank you so much.

bblanchon, 2021-07-31:
Hi @bhcuong2008,

You must use the capacity required by the largest message.
A `JsonDocument` is a temporary buffer only used during deserialization, so it should not impact the rest of your application.

See also:

* [How to determine the capacity of the JsonDocument?](https://arduinojson.org/v6/how-to/determine-the-capacity-of-the-jsondocument/)

Best regards,
Benoit

bhcuong2008, 2021-08-01:
Thank @bblanchon .

If a string buffer created from serializeMsgPack, could we estimate how much size of new buffer required  when doing deserializeMsgPack based on that source buffer?

bhcuong2008, 2021-08-01:
I also have a question related to this. For document deserialization from char* (not copy mode), does memoryUsage() include some strings from that char*?

For example, a char buf: {"a":1, "b":2} on ESP32 platform. Based on https://arduinojson.org/v6/assistant/, I have

When deserializeJson(doc, (char*)buf), minimum capacity for the JsonDocument is 32.

When deserializeJson(doc, (const char*)buf), minimum capacity for the JsonDocument is 36.

In both cases, is doc.memoryUsage() 36?

bblanchon, 2021-08-02:
Hi @bhcuong2008,

I just ran the following program on ESP32, and it confirms that `memoryUsage()` returns 36 and 32:

```c++
StaticJsonDocument<64> doc;

char json[] = "{\"a\":1,\"b\":2}";
deserializeJson(doc,(const char*)json);
Serial.println(doc.memoryUsage());  // 36
  
deserializeJson(doc,(char*)json);
Serial.println(doc.memoryUsage());  // 32
```

Please provide a repro so I can understand why you get 36 for both.

Best regards,
Benoit

bhcuong2008, 2021-08-02:
Ah, it's my question. Not my result. 
Your test helps answer my question. However I expect memoryUsage having parameters to return its own usage buffer size or total.

I use total size to define a new doc when I deserialize it again instead of defining a large buffer, such as 4KB. It helps reduce heap memory runtime while processing data. Its life time may last hundreds of ms. So reducing doc buffer runtime is a must for my app.

bhcuong2008, 2021-08-02:
I give an example to help you got my idea.
Here is a message got from MQTT. After parsing topic, validate some things, then save key "parameters" for future processing. 

const char* payload = {"parameters":{"data":[[20,0,{"_cntButton":8}]],"quiet":1},"_conn":1}

At the 1st time, I must allocate document buffer as large as possible to do pre-processing, like this:
DynamicJsonDocument doc(4096);
deserializeJson(doc, (char*)payload);
I use char* for faster time, and saving RAM.

Then I do pre-processing, and decide to keep doc["parameters"] for future (asynchronous) processing. At this time, I save doc["parameters"] size. If using deserialize char*, it will be 112 bytes. If deserialize with const char*, it will be 134 bytes.

In my case, using char*, it will be 112 bytes. I save this number, and serialize doc["parameters"] to string params.
JsonObject jsParams = doc["parameters"].as<JsonObject>();
uint16_t paramsSz = jsParams.memoryUsage();   // 112
std::string params;
serializeMsgPack(doc["parameters"], params);

Now, it's time to process this, I need to deserialize it. Previous doc already destroyed. Do not care it.
DynamicJsonDocument doc(paramsSz);   // if use fixed 4096 here, it will be wasteful. And maybe not enough memory if there are more waiting message.
deserializeMsgPack(doc, params);

In this case, it will be not enough buffer. Deserialize failed. Its required buffer must be 134.

So that is my case. I need memoryUsage() having an option to get total size when deserialize with char*.

bhcuong2008, 2021-08-02:
I even expect ArduinoJson having a runtime helper function to scan input string to get total size before allocation, like this.

```
size_t docSize = getBufferSizeJson(payload);
DynamicJsonDocument doc(docSize);
deserializeJson(doc, (char*)payload);
```

bblanchon, 2021-08-04:
Can you explain "if use fixed 4096 here, it will be wasteful. And maybe not enough memory if there are more waiting message."?

bhcuong2008, 2021-08-04:
If we dont know doc buffer size in advance, then we must allocate as much as possible, in my case it's 4KB. 

`DynamicJsonDocument doc(4KB);`

The doc's lifetime is up to several hundreds of ms (worst case in my case) due to processing command async with MQTT callback. 
In MQTT callback, I only pre-process message to validate it, then save message in queues at global scope. 

Then in main loop, I will process that command by deserializing message again, and that doc lifetime will last until command done, and the message will be removed.

My device can accept up to tens of messages in command queues waiting for async processing.
For example, one command is device/set, other command is device/get,...Some commands will last long time because it may save to flash, or read from flash, or waiting IO from UART, I2C,..

So if using doc with exactly size will help save RAM. With ten commands in processing, it will take 40KB for ten JsonDocument. While if known size, ex 512 bytes, then the device could handle up to 80 commands.

bhcuong2008, 2021-08-04:
Currently I use deserialize with const char* in pre-proccessing phase to get exactly buffer usage size. But using const char* (copy mode) takes time and RAM although processing time just several ms in this pre-processing phase.

bblanchon, 2021-08-05:
I suggest that you store the original (i.e., unparsed) JSON document in the queue: it's much smaller than the equivalent `DynamicJsonDocument`.
 Then, you can deserialize in the main loop with a single `DynamicJsonDocument` of 4K.

BTW, I don't know what library you use for MQTT, but if I take PubSubClient as an example, it calls the callback function from the main loop, so it cannot receive any new message while the main loop is busy.

bhcuong2008, 2021-08-05:
Currently I store message as string from serializeMsgPack when receiving a message from MQTT callback (PubSubClient), stored it in queue, together with buffer size (thanks to memoryUsage()), and exit as soon as possible.

When it's time to process the queue, I deserializeMsgPack it with exact buffer size, instead of max 4KB (I limit this). Some commands will be finished in this cycle, and the doc is released immediately at that time.

But for some commands, I need to process async. Everytime it's run, I deserialize it again.

I want to save RAM and faster time (use not copy mode) at pre-process phase.
Also I want to hold full json doc, so that I dont need to deserialize it everytime I process that command. And when done finally, I free that document.

Anyway, this is not bug. Just it arises from my needs of better performance.
Thank you for your help.

bblanchon, 2021-08-07:
Your use-case isn't quite aligned with the design of the library.
Again, `JsonDocument` is supposed to be a temporary buffer that your use during the (de-)serialization.
In your case, you're using `JsonDocument` as a medium to store some data and pass it via a queue.
ArduinoJson 7 will permit this use case but with a significant overhead ([see what happened with 6.6](https://arduinojson.org/news/2018/12/07/version-6-7-0/))

As a workaround, I recommend using a 4KB memory pool and shrink it with [`shrinkToFit()`](https://arduinojson.org/v6/api/basicjsondocument/shrinktofit/).

Also, if you want to improve any kind of metric, especially performance, you must measure first.
I'm really not convinced that the gymnastic of passing a `JsonDocument` around is faster than copying the original string to the queue. It's up to you to figure this out.

bhcuong2008, 2021-08-07:
Thank you @bblanchon , I've just looked at your document. I got it. I will not use document for multi-phases processing. I will use the previous way as above. I store messages in string in MsgPack format. Then every time I process that command, I will deserialize it.

However could you tell me how to measure the whole document memory usage, even in case of char* deserialization? I think it's possible because in non-copy mode (char*) deserialization, it stores string pointers. When counting, we can strlen that strings too. This will help my pre-processing phase, deserialize in non-copy mode on MQTT receive buffer for faster and saving RAM.

bblanchon, 2021-08-10:
I'm sorry, @bhcuong2008, but there is no way to measure the required capacity without parsing the document first.

I'm not going to add such function to the library because it would:

1. require significant work from my side with very little return (if any)
2. considerably slow down your program (two deserialization instead of one)
3. considerably increase the size of your program (two instantiations of the parser)
4. potentially increase heap fragmentation (block of different sizes)
5. not be applicable to stream input (the vast majority of use cases)
6. go against the design of the library.

You think fixed memory allocation is a waste of RAM, but I can assure you that it's much better than fragmenting the heap.

closed 2021-08-21 by bblanchon