bblanchon/ArduinoJson #1620: Serialization as a callback?

bmweller, 2021-08-01:
I'm new to working with Json on the arduino platform.

I'm working on a project which integrates an AVR and a Tasmota ESP8266.

Periodically the AVR sends telemetry to Tasmota device over Serial.

I'd like to structure the content being serialized with a backed Json doc from this library while at the same time use the TasmotaClient library command response features.

The [TasmotaClient](https://github.com/andrethomas/TasmotaClient/tree/master/src) has a few entry points into the Serial stream write (solicited and unsolicited) such that the write is wrapped in a command tag.

```c++
void TasmotaClient::SendTele(char *data)
{
  SendCommand(CMND_PUBLISH_TELE, strlen(data));
  serial->write(char(PARAM_DATA_START));
  for (uint8_t idx = 0; idx < strlen(data); idx++) {
	  serial->write(data[idx]);
  }
  serial->write(char(PARAM_DATA_END));
}
```

I'd like to avoid marshalling the doc to the character array first, is there a more elegant way of using serializeJson to yield a character stream?

bblanchon, 2021-08-02:
Hi @bmweller,

I don't have any experience with Tasmota, and I'm not sure to understand your message entirely, but I can answer the followings questions.

> I'd like to avoid marshalling the doc to the character array first

Yes, you can.

> is there a more elegant way of using serializeJson to yield a character stream?

[`serializeJson()`](https://arduinojson.org/v6/api/json/serializejson/) can write directly to a `Print`, `Stream`, `Client`, and `std::ostream`.
You can even add support for new streams via [custom writers](https://arduinojson.org/news/2019/11/01/version-6-13-0/).

However, it imposes that you serialize the *complete* JSON document; it cannot incrementally produce the output
(nor can [`deserializeJson()`](https://arduinojson.org/v6/api/json/deserializejson/) read incrementally, the input must be complete too).

Best regards,
Benoit

bmweller, 2021-08-02:
thanks, Benoit, I appreciate the suggestion, will now close this question.

closed 2021-08-02 by bmweller