bblanchon/ArduinoJson #1621: ArduinoJson.h Compile error <call of overloaded 'beginMessage(const char [8], size_t&)' is ambiguous>

mackrasp, 2021-08-01:
<Arduino 1.8.15 IDE compiler error running on macOS 10.14.6 Macmini
Target device Arduino Mega 2560

call of overloaded 'beginMessage(const char [8], size_t&)' is ambiguous

/Users/mack/Documents/Arduino/Mega_Eth_MQTT_Ard_J_Pub_v1_2/Mega_Eth_MQTT_Ard_J_Pub_v1_2.ino: In function 'void loop()':
Mega_Eth_MQTT_Ard_J_Pub_v1_2:112:49: error: call of overloaded 'beginMessage(const char [8], size_t&)' is ambiguous
   mqttClient.beginMessage("myTopic", payloadSize);
                                                                                      ^
In file included from /Users/mack/Documents/Arduino/libraries/ArduinoMqttClient/src/ArduinoMqttClient.h:23:0,

I have copied your code from but it errors on compilation.
https://arduinojson.org/v6/how-to/use-arduinojson-with-arduinomqttclient/
If you want to specify the size to beginMessage(), you could use measureJson(), but it’s faster to use a temporary buffer, like that:

`char payload[64];
size_t payloadSize = serializeJson(doc, payload);

mqttClient.beginMessage(topic, payloadSize);
mqttClient.print(payload);
mqttClient.endMessage();

my code in void loop()
void loop() {
mqttClient.poll();
currentMillis = millis();
if (currentMillis - previousMillis >= interval) {
    previousMillis = currentMillis;
  
  StaticJsonDocument<256> doc;
  doc["sensor"] = "gps";
  doc["time"] = 1351824120;

  // Add an array.
  JsonArray data = doc.createNestedArray("data");
  data.add(48.756080);
  data.add(2.302038);

  // Generate the minified JSON and send it to the Serial port.

  char payload[64];
  size_t payloadSize = serializeJson(doc, payload);

  mqttClient.beginMessage("myTopic", payloadSize);    //this line errors
  mqttClient.print(payload);
  mqttClient.endMessage();      
  }  
} //end loop
`

!--
Before opening an issue, please read the FAQ:
https://arduinojson.org/faq/

Please provide all the relevant information:
* good title
* short description of the problem
* target platform
* compiler model and version
* MVCE (https://stackoverflow.com/help/mcve)
* compiler output

Good questions get fast answers!
-->

bblanchon, 2021-08-02:
Hi @mackrasp,

Thank you very much for reporting this error 👍 
To fix it, you must cast `payloadSize` to `unsigned long`, like so:

```c++
mqttClient.beginMessage("myTopic", (unsigned long)payloadSize);
```

I updated the documentation accordingly.

Best regards,
Benoit

mackrasp, 2021-08-03:
Hi Benoit,
All OK now. Thanks
What is the difference between the Stream method and the payloadSize method for serializeJson()?
I am intending to use ArduinoMqttClient.h and ArduinoJson.h to publish/subscribe over a local 192.168.x.x network using ethernet and wifi. The data will be the topic name, a time stamp long integer and a couple of long integers. With a publish interval of 5minutes. Which do you suggest?

Regards
Mack

Reply to: bblanchon/ArduinoJson ***@***.***>
Date: Monday, 2 August 2021 at 3:45 pm
To: bblanchon/ArduinoJson ***@***.***>
Cc: Mack Richards ***@***.***>, Mention ***@***.***>

bblanchon, 2021-08-03:
Hi Mack,

I recommend that you serialize to a temporary buffer because the stream method is pretty slow in this case.
For an explanation, see "Can we avoid the temporary buffer?" in [How to use ArduinoJson with PubSubClient?](https://arduinojson.org/v6/how-to/use-arduinojson-with-pubsubclient/)

Best regards
Benoit.

bblanchon, 2021-08-15:
Mack, did you solve your problem?

mackrasp, 2021-08-15:
Hi Benoit,
My project is not yet completed, although I have successfully test published and subscribed using ArduinoMqttClient and ArduinoJson using the temporary buffer method.
Thanks again for your assistance and suggestions.
Regards
Mack

Date: Sunday, 15 August 2021 at 4:19 pm

bblanchon, 2021-08-16:
Thank you for the update, Mack.
Good luck with your project.

closed 2021-08-16 by bblanchon