bblanchon/ArduinoJson #1622: Compilation error introduced since 6.18.3

labodj, 2021-08-02:
Hi @bblanchon,
I updated this library from 6.18.2 to 6.18.3 and I'm experiencing compilation errors, everything was fine in 6.18.2

I'm compiling using avr-gcc 11.1.0 with c++20 features for an Arduino mega 2560

These are the errors:
```txt
/ArduinoJson/src/ArduinoJson/Strings/Adapters/ConstRamStringAdapter.hpp:48:32: error: expected unqualified-id before 'const'
/ArduinoJson/src/ArduinoJson/Strings/Adapters/ConstRamStringAdapter.hpp:48:32: error: expected ')' before 'const'
/ArduinoJson/src/ArduinoJson/Strings/Adapters/ArduinoStringAdapter.hpp:48:37: error: expected unqualified-id before 'const'
/ArduinoJson/src/ArduinoJson/Strings/Adapters/ArduinoStringAdapter.hpp:48:37: error: expected ')' before 'const'
```

The first error is located here:
```cpp
template <int N>
class StringAdapter<const char[N]> : public StringAdapter<const char*> {
 public:
  StringAdapter<const char[N]>(const char* s) : StringAdapter<const char*>(s) {}
};
```
The second one is located here:
```cpp
template <>
class StringAdapter< ::StringSumHelper> : public StringAdapter< ::String> {
 public:
  StringAdapter< ::StringSumHelper>(const ::String& s)
      : StringAdapter< ::String>(s) {}
};
```

Since 6.18.2 the 2 files above have been modified with these commits:
5790f3c8f728036c24999cd3c3639ddb1fed41b7
6632fa8da1316c002cf8bcc3fde804ebd0be5bd4

bblanchon, 2021-08-03:
Hi @labodj,

Thank you very much for reporting this bug 👍 
It should be fixed in the `6.x` branch.

Best regards,
Benoit

closed 2021-08-03 by bblanchon

bblanchon, 2021-09-06:
The fix was published in [ArduinoJson 6.18.4](https://github.com/bblanchon/ArduinoJson/releases/tag/v6.18.4).

commit f570fe8c (2021-08-03): Fixed error `expected unqualified-id` on GCC 11 (fixes #1622)