bblanchon/ArduinoJson #1623: Unable to read deserialized nested json arrays on ESP32

t3chn1ca, 2021-08-06:
First of all thanks for the great library.

Some details of the setup:
Platform: ESP32
Arduino Json version:  6.x latest.

Issue: 
Issue being observed is deserialized JSON fails to read the nested "data" section of the below JSON, 
but does correctly read the "day" field ( which is not nested ).
```
[{
	"data": [{
		"time": "10",
		"revs": 1
	}, {
		"time": "11",
		"revs": 2
	}],
	"day": "Mon"
}, {
	"data": [{
		"time": "10",
		"revs": 1
	}, {
		"time": "11",
		"revs": 2
	}],
	"day": "Tue"
}]
```
Sample source ( snippet uses ArduinoJson Assistant generated sample ) :
```c++
#include "ArduinoJSON.h"
void setup()
{
    String input = "[{\"data\":[{\"time\":\"10\",\"revs\":1},{\"time\":\"11\",\"revs\":2},{\"time\":\"12\",\"revs\":3}],\"day\":\"Mon\"},{\"data\":[{\"time\":\"10\",\"revs\":1},{\"time\":\"11\",\"revs\":2}],\"day\":\"Tue\"}]";
    DynamicJsonDocument doc(1536);
    Serial.begin(115200);
    Serial.print("Doc capacity: ");    
    Serial.println(doc.capacity());    

    DeserializationError error = deserializeJson(doc, input);

    //There is no error seen here, removed error check for smaller code sample

    for (JsonObject elem : doc.as<JsonArray>())
    {
        Serial.println("===================================");
        for (JsonObject elem : elem["data"].as<JsonArray>())
        {
            Serial.println("Nested  data");  // <======================== Does not come into this section
            const char *time = elem["time"]; // "10", "11", "12", "13", "18", "15", "20"
            Serial.println(time);
            int revs = elem["revs"];         // 1, 2, 3, 4, 1, 1, 1
            Serial.println(revs);
        }
        const char *day = elem["day"]; // "Mon", "Tue", "Wed", "Thu", "Fri", "Sat", "Sun"
        Serial.println(day);  // <======================== this field is correctly read 
    }
}
```
Output obtained on ESP32 Serial is  as below, missing the nested  "time" and "revs" fields:
```
Doc capacity: 1536
===================================
Day: Mon
===================================
Day: Tue
```

bblanchon, 2021-08-06:
Hi @t3chn1ca,

Thank you very much for reporting this issue 👍 

The problem comes from the `elem` variable in the inner loop which shadows the `elem` variable of the inner loop.
Renaming one of the two variables fixes the issue as you can see in the following demo:
https://wandbox.org/permlink/2oP1KrSfSthKsIci

I'll modify the ArduinoJson Assistant so that it doesn't reuse the same variable name.

Best regards,
Benoit

bblanchon, 2021-08-06:
Done! The Assistant now uses the names `item` and `data_item` for these two variables.

closed 2021-08-06 by bblanchon

t3chn1ca, 2021-08-06:
Thanks for the quick response!