bblanchon/ArduinoJson #1624: Any example of deserializemsgpack() with serial read?

BRoboticsSHADO, 2021-08-06:
Hi

Do you have any example of using deserializemsgpack() with serial.read() ? How to identify when the incoming msgpack of the json object ends?

For example, the following shows how to read the incoming byte of String, it use carriage return or EOL to differentiate the complete of receiving a String
```
  while (Serial.available()) {
    if (countDataPC >= maxCharNum) {
      serialDataPC = "";
      boolDataPC = false;
      countDataPC = 0;
      break;
    }
    char c = SerialPC.read();
    serialDataPC += c;
    if (c=='\r' || c=='\n') {
      boolDataPC = true;
      break;
    } 
    boolDataPC = false;
    countDataPC++;
  }

  if (boolDataPC) {
    DynamicJsonDocument doc(1024);
    auto error = deserializeMsgPack(doc, serialDataPC);
    JsonObject readJson = doc.as<JsonObject>();
    if (!error) {
      readDataJSON(readJson);
      writeDataJSON(SerialPC);
    }
    serialDataPC = "";
    boolDataPC = false;
    countDataPC = 0;
  }
```

How shall i identify the incoming msgpack of a json object already ended?

Best,
Samuel

bblanchon, 2021-08-07:
Hi Samuel,

You can pass `Serial` directly as the input of [`deserializeMsgPack()`](https://arduinojson.org/v6/api/msgpack/deserializemsgpack/); it will stop reading at the end of the object.

See also:

* [How to do serial communication between two boards?](https://arduinojson.org/v6/how-to/do-serial-communication-between-two-boards/) (the same principles apply, only the names of the functions change)

Best regards,
Benoit

BRoboticsSHADO, 2021-08-11:
Hi @bblanchon 

Thank you for the suggestions. It helps a lot. Now i can get deserializeMsgPack to work normally.

Best,
Samuel

closed 2021-08-11 by BRoboticsSHADO

bblanchon, 2021-08-12:
You're welcome, Samuel!
Thank you for using ArduinoJson.
Don't forget to [cast a star](https://github.com/bblanchon/ArduinoJson/stargazers) to support the project :wink: