bblanchon/ArduinoJson #1625: JsonArray document is thread-safe?

bhcuong2008, 2021-08-07:
Hi,

My ESP32 application has 2 sides: Arduino app in C++ part (in mainloop thread), and MicroPython (in other thread) as library.

I transfer message data to MicroPython thread using pre-allocate JsonDocument JsonArray as FIFO. App side will push to the FIFO. MicroPython side will get element 0 from FIFO. I clear the whole document when the FIFO is empty in app side, as recommend from ArduinoJson API document.

So for these operations, is it thread-safe? 

Thank you so much,
Cuong.

bblanchon, 2021-08-08:
Hi Cuong,

Each `JsonDocument` must be accessed by one thread at a time, including all the `JsonArray`s pointing to it.

I'm assuming that you're transferring JSON strings between the two threads, so `JsonDocument` is not affected by the concurrency.
Instead, if you're exchanging a `DynamicJsonDocument` between the threads (which is not what the library was designed for; ArduinoJson is a *serialization* library), it will be safe as long as you only use it from one thread at a time.

Regarding my recommendation about "clearing" the `JsonDocument`, it really doesn't sound like me.
Can you show me where you saw that?

Best regards,
Benoit

bhcuong2008, 2021-08-14:
Hi @bblanchon , thank you.

About clearing JsonDocument, pls look at [https://arduinojson.org/v6/api/jsonarray/clear/](url)

> Causes memory leaks ⚠️
> Because JsonDocument contains a monotonic allocator, this function cannot release the memory associated with the removed key-value pair.
> Consequently, you cannot call this function in a loop; otherwise, the JsonDocument will overflow.
> 
> Prefer JsonDocument::clear() which doesn’t leak memory.

So I follow that guide, when checking JsonArray empty, I also clear JsonDocument too.

About my use case of JsonDocument as FIFO, I use it together with Queue FreeRTOS. Each item in queue contains just meta data. And actual data, ex topic, payload, I push it to JsonArray. I will use mutex to protect this when doing clearing item in JsonArray, and clearing JsonDocument.

bhcuong2008, 2021-08-14:
One more concern is that, if one thread is removing item 0 in JsonArray, and another thread is pushing (`JsonArray.add()`) a new item to JsonArray, then is there any race condition?

bblanchon, 2021-08-15:
I strongly recommend against using `JsonDocument` as a FIFO because the monotonic allocator cannot release memory, and the memory pool will quickly overflow.

ArduinoJson is a *serialization* library, not a data structure library.
Instead, you should use containers from the Standard C++ library.

bhcuong2008, 2021-08-15:
Thank you @bblanchon. 
I already run many tests without any problems. It's smooth.

bhcuong2008, 2021-08-16:
Based on your recommendation, I'm trying to implement it using std::vector, together with mutex. Not using FreeRTOS queue, and JsonDocument any more.

Thank you so much.

bblanchon, 2021-08-16:
`std::deque` is probably more appropriate for a FIFO.
Good luck with your project, @bhcuong2008.

closed 2021-08-16 by bblanchon

bhcuong2008, 2021-08-16:
I've just looked at it. deque is really great in this use case. Thank you very much.