bblanchon/ArduinoJson #1627: Any example to Serialize and deserialize on the same serial?

Bingsitfx, 2021-08-09:
Hello, i have a problem with my program.
I want to communicate between Mega and ESP8266. Im using Robotdyn Arduino Mega wifi [ https://robotdyn.com/mega-wifi-r3-atmega2560-esp8266-flash-32mb-usb-ttl-ch340g-micro-usb.html](url).

This is my Serialize output on Arduino Mega
[image]

It supposed to look like this `{"1":62.1268,"2":61.79792,"3":62.10856,"4":63.89173,"5":61.09426,"6":58.33428,"rata":61.55893}`

My serialize code (arduino) : 
```
 DynamicJsonDocument data1(512);
 data1["1"] = sensorValue1;
 data1["2"] = sensorValue2; 
 data1["3"] = sensorValue3;
 data1["4"] = sensorValue4;
 data1["5"] = sensorValue5;
 data1["6"] = sensorValue6;
 data1["rata"] = rata;
 serializeJson(data1, Serial3);
```

My deserialize code (esp8266) : 
```
 DynamicJsonDocument data1(512);
 DeserializationError err = deserializeJson(data1, Serial);
 if (err == DeserializationError::Ok) {
     sensor1 = data1["1"];
     sensor2 = data1["2"];
     sensor3 = data1["3"];
     sensor4 = data1["4"];
     sensor5 = data1["5"];
     sensor6 = data1["6"];
     rata = data1["rata"];
```

My question is i'm doing something wrong?
How to solve this?
```
note : serialize ESP8266 => deserialize arduino is working properly on the same serial.
       Serial3 Arduino connected to Seral ESP8266
```
Thankyou

bblanchon, 2021-08-10:
Hi @Bingsitfx,

I'm sorry, I don't understand your problem; can you be more specific?
Did you try the [ArduinoJson Troubleshooter](https://arduinojson.org/v6/troubleshooter/)?
Did you read [How to do serial communication between two boards?](https://arduinojson.org/v6/how-to/do-serial-communication-between-two-boards/)
This doesn't look like a JSON problem; did you contact RobotDyn support?

Best regards,
Benoit

dpetican, 2021-08-17:
@Bingsitfx I've been doing something similar for years (with version 5), a Mega, and an ESP8266.  It works great with caveats not directly related to ArduinoJson.  BTW the link to the board doesn't seem to work with me.  But I have looked at similar boards with WiFi and they can be tricky.

Still it seems to me that you do not have a general communications problem.  Just to clarify:

1) Is it working one way but not the other or both.  If one way which way?
2) Its looks like the json is formatted properly.  Is it just the numbers that are wrong?

Bingsitfx, 2021-08-18:
> Hi @Bingsitfx,
>
> I'm sorry, I don't understand your problem; can you be more specific?
> Did you try the [ArduinoJson Troubleshooter](https://arduinojson.org/v6/troubleshooter/)?
> Did you read [How to [...]
I want to receive & send Json data on the same serial port. Robotdyn have switch for connecting port between ATmega2560 and ESP8266, the switch is on Serial3. So atmega2560 serial3 connected to esp8266 serial0. I managed to send json data from esp to arduino but can't send json data from arduino to esp.

I've tried various ways but nothing works so i taught it is impossible. Yesterday I tried lowering the baudrate and it worked with 57600 bps. In conclusion my problem is only in the hardware, not all esp can send data at 115200 baudrate, it depends on manufactures.

Bingsitfx, 2021-08-18:
> @Bingsitfx I've been doing something similar for years (with version 5), a Mega, and an ESP8266. It works great with caveats not directly related to ArduinoJson. BTW the link to the board doesn't [...]

It is working one way, esp to arduino. I tried lowering the baudrate to 57600 and now is working both way.

dpetican, 2021-08-18:
> It is working one way, esp to arduino. I tried lowering the baudrate to 57600 and now is working both way.

Flakey hardware may indeed be the problem, but perhaps its another one as well.  I've just recently learned that ArduinoJson deserializes input one byte at a time.  What this means is the hardware receive buffer can be overrun if you send too large messages too quickly.  But your json format didn't look corrupted just the numbers.  Anyway, in case your interested see this issue: https://github.com/bblanchon/ArduinoStreamUtils/issues/15#issue-972932300

Given the size of your json consider (like I am) using the messagepack methods (deserializeMsgPack serializeMsgPack) to reduce the number of bytes sent and StreamUtils to consume the hardware buffer faster.  Really depends on the time between messages.  BTW I think that slower baudrates can be better for sending larger messages as the hardware buffer on the receiving end does not fill up as fast.  As I eluded to in the other issue this is really all down to managing flow without hardware or software flow control 😀

bblanchon, 2021-08-19:
@Bingsitfx, when using one serial to print debug information and a second serial port for communication, it is crucial that the speed of the « debug » port is significantly faster than the « communication » port; otherwise, the program may read too slowly from the « comminucation » port and drop some bytes (which usually leads to `InvalidInput`).
I recommend running the debug port at least ten times faster than the communication port.

Bingsitfx, 2021-08-19:
> > It is working one way, esp to arduino. I tried lowering the baudrate to 57600 and now is working both way.
> 
> Flakey hardware may indeed be the problem, but perhaps its another one as well. I've just recently learned that ArduinoJson deserializes input one byte at a time. What this means is the hardware receive buffer can be overrun if you send too large messages too quickly. But your json format didn't look corrupted just the numbers. Anyway, in case your interested see this issue: [bblanchon/ArduinoStreamUtils#15 (comment)](https://github.com/bblanchon/ArduinoStreamUtils/issues/15#issue-972932300)
> 
> Given the size of your json consider (like I am) using the messagepack methods (deserializeMsgPack serializeMsgPack) to reduce the number of bytes sent ~and StreamUtils to consume the hardware buffer faster.~ Really depends on the time between messages. BTW I think that slower baudrates can be better for sending larger messages as the hardware buffer on the receiving end does not fill up as fast. As I eluded to in the other issue this is really all down to managing flow without hardware or software flow control 😀

I think so, i try 115200 baudrate that send "<200>" Its success without an error, try larger error occured 😆. Also dont forget to use bidirectional logic level converter if you want to send data with serial port on the diferent voltage like 3,3 and 5V.

Bingsitfx, 2021-08-19:
> @Bingsitfx, when using one serial to print debug information and a second serial port for communication, it is crucial that the speed of the « debug » port is significantly faster than the « [...]

Thankyou for your advice, next time i will try it😃
I think you should share some example how to determine how fast is the baudrate to send json data, or perhaps you can add baudrate  calculations on arduinonson assistant. it's very helpful for beginners like me

bblanchon, 2021-08-20:
I already wrote about this issue here:

* [How to do serial communication between two boards?](https://arduinojson.org/v6/how-to/do-serial-communication-between-two-boards/)
* [ArduinoJson Troubleshooter](https://arduinojson.org/v6/troubleshooter/#runtime/deserialization/invalidinput/serial/hardware/voltage-ok)

Please let me know if there are other places in the documentation where you expect to find this information.

Bingsitfx, 2021-08-20:
[image]

I think if you add a baudrate recommendation on this web, it will be very helpful.

dpetican, 2021-08-20:
@Bingsitfx see the last post here: [Relationship between baud rate and Bps?](https://www.microchip.com/forums/m405110.aspx).  In my case I'm allowing for 256 bytes to be transmitted so:

10bits/9600 Baud = 1041.66666666 us/byte
256bytes x 1041.66666666 us/byte = ~267ms

But then I have to factor in the hardware transmit buffer on the Mega, hardware receive buffer on the ESP8266 both of which I've set at 128bytes.  Although its not quantifiable they improve the effective bytes/sec.

Now you have to add overhead which is the time it takes to deserialize the message on the ESP8266.  I'm not concerned with how long it takes to serialize the message on the Mega as I only want to prevent receive buffer overflows on the ESP8266.

How do you calculate the last two factors?  You can't... easily.  I determined by experimentation that in my application for 256bytes max an inter-message delay of 300ms works at 9600baud.  In the next version I'm going to make the delay adaptive using a fixed overhead and the formula above. Also, I'm switching from an ESP8266 to an ESP32 which means the deserializer will consume bytes faster from the hardware buffer.

closed 2021-08-20 by bblanchon