bblanchon/ArduinoJson #1628: Exclude key using filter

dmadison, 2021-08-13:
Hello,

I'm wondering if it's possible to exclude a specific key in a filter rather than explicitly including all of the others.

For context, I have a JSON response I'm trying to parse that has a large array of strings I don't need. I'd like to keep the other ~15 keys, but it's a pain to keep track of every other key and mark each of those keys explicitly as "true" in the filter rather than marking one key as "false".

Here's a minimal sample of the JSON data:

```cpp
{
  "data": [
    {
      "id": "123456789",
      "name": "John",
      "started_at": "2021-08-12T19:54:18Z",
      "language": "en",
      "tag_ids": [
        "ffb3744956f39b69bafd777c75efe3d3",
        "dbbc7932c63c7a3776a7bc0c6f7696f1",
        "cbc69d36b95b3d083fa78707bf4d67df",
        "a3d0a2b834bda602fdd962f0f84c6210",
        "26677562c8d68ce512f0ddc4d8390c69",
        "dc816b81f3c4524a42838698859efbb6",
        "f145b3b9d0567f41e658aa19d1c3fc65",
        "2f173d6ef410aa016a40ffba3b11bf88",
        "6f389ef6e9c811144b5efc63abbf83c9",
        "fa4bb4e69607560b6007710f175dd302"
      ]
    }
  ]
}
```

And here's my current filter, using a wildcard:

```
{
  "data": [
    {
      "*": true,
      "tag_ids": false
    }
  ]
}
```

The Assistant successfully filters out the "tag_ids" key, but the library on platform does not. I did not find any documentation using "false" values, so my guess is that this is not currently a supported feature.

I'm testing an ESP8266 (Wemos D1 Mini Lite) using Arduino 1.8.13. Here's the current C++ code:

```cpp
StaticJsonDocument<96> filter;
filter["data"][0]["*"] = true;
filter["data"][0]["tag_ids"] = false;
StaticJsonDocument<2000> doc;
deserializeJson(doc, json, DeserializationOption::Filter(filter));
```

Thanks for any advice!

bblanchon, 2021-08-13:
Hi @dmadison,

Thank you very much for reporting this issue 👍 
I didn't think of exclusion when I added the wildcard, but I think it's a valid use case.
I'll fix the library to match the behavior of the Assistant.

Best regards,
Benoit

closed 2021-08-13 by bblanchon

dmadison, 2021-08-13:
Thank you so much for the fast response and solution @bblanchon! I just tested and it works great. Cheers!

bblanchon, 2021-08-14:
You're welcome, @dmadison!
Thank you for using ArduinoJson.

bblanchon, 2021-09-06:
The fix was published in [ArduinoJson 6.18.4](https://github.com/bblanchon/ArduinoJson/releases/tag/v6.18.4).

commit 2c3ae1c1 (2021-08-13): Filter: exact match takes precedence over wildcard (fixes #1628)